flashrom/flashrom #23: MX25L25635F Support

daniel-rome, 2017-09-29:
I have seen that there is no support for the MX25L25635F flash. In fact, I have seen a reference to it within the "flashchips.h" file (MACRONIX_MX25L25635F), but there is no references to it within the flashchips array ("flashchips.c" file).

Is it still supported?

stefanct, 2017-10-08:
It is not supported **yet** since there is no support for flash chips larger than 128 MBit (yet). The ID in the header is just a reminder for the future.

Keno, 2017-10-08:
I've been able to successfully program this chip with the changeset at https://review.coreboot.org/#/c/19855/.

Keno, 2017-10-15:
Looks like that changeset was just merged (5bdb87e61f154524f37e249fa6ddae893840b9e5).

dhendrix, 2018-02-12:
I think this issue has been resolved. We merged 4BA support and the MX25L25635F is one of the supported chips.

closed 2018-02-12 by dhendrix