flashrom/flashrom #29: Does not work without spispeed= on RPi

probonopd, 2017-12-21:
`flashrom v0.9.9-r1954 on Linux 4.9.59-v7+ (armv7l)` (the one coming with the most recent Raspbian) does not seem to work without `spispeed=...` on RPi:

```
# Does NOT work
pi@raspberrypi:~ $ sudo flashrom -p linux_spi:dev=/dev/spidev0.0 -r file.rom
(...)
Calibrating delay loop... OK.
Found Generic flash chip "unknown SPI chip (RDID)" (0 kB, SPI) on linux_spi.
```

Turns out it does work when specifying `spispeed= `:

```
# Works - SLOW reading
pi@raspberrypi:~ $ sudo flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=512 -r slow.rom

# Works - FAST reading
pi@raspberrypi:~ $ sudo flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=30000 -r fast.rom

# Check whether both are identical
pi@raspberrypi:~ $ diff slow.rom fast.rom
```

iomonad, 2018-02-15:
Same here. It work only if specified otherwise the chip become unknown.

probonopd, 2018-02-16:
Thanks for confirming @iomonad - I think this should be fixed by hardcoding a default value. It took me some time to find this out.

kfowlks, 2018-02-19:
@iomonad I can confirm this as well. I had the same issue with a pi-zero running a  4.9.59 Linux kernel and the most recent stable release of flashrom built from git.

dhendrix, 2018-02-20:
On my rpi3 it seems to work without the spispeed parameter but very slowly. The devicetree shows spi-max-frequency set to 500KHz.

A hardcoded default is not a good solution since RPi is not the only user of the linux_spi driver and poor quality connection to the chip can require slowing down the speed as well. It might be worth asking the Raspberry Pi foundation to increase spi-max-frequency, but I don't expect that to land since they probably have other use cases in mind.

At the very least perhaps we can make flashrom print a message to the user to adjust spispeed if the chip ID is not found.

docpyro, 2018-02-24:
Your comment fixes the issue I was having for days. Thank You Very Much!!!!!!!

wareck, 2018-04-23:
Hello
i've just tested this software.
For me on a RPI3 it's better to use sudo flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=20000
It's a good compromize to speed and less error

jal2, 2018-05-13:
Same problem here with flashrom version p1.0-74-g2568357 on a RPi 1 model B. Kernel is 4.14.34+, distribution Raspbian GNU/Linux 9.

rnamdos, 2018-08-21:
I was able to finally probe my Macronix chip once (and only once) by setting the speed to 20000... cant get it to work again. I'm using Pi zero

saraazizi76, 2019-07-28:
sorry. i have a question.some commands are listed in man page (8) for reading, writing, ... .where this commands should be run? in terminal of ubuntu? how can i run flashrom software???

docpyro, 2019-08-16:
yes terminal if you have flashrom downloaded and setup as discussed
elsewhere

saraazizi76, 2019-08-18:
> yes terminal if you have flashrom downloaded and setup as discussed elsewhere
> […](#)

tnx
i have another problem now. i wanna use ft2232_spi programmer , but when i connect this programmer to my laptop and run this command "./flashrom -r ../output_EEPROM -p ft2232_spi" in the terminal i got this error"Unable to open FTDI device: -3 (device not found).
Error: Programmer initialization failed."
Do you know why?

probonopd, 2019-08-18:
Maybe try with `sudo`?