flashrom/flashrom #31: FT(2)232H Dual/Quad SPI

c0d3z3r0, 2018-01-23:
There are SPI flash chips like W25Q128FV that support dual/quad SPI (using 2/4 inputs/outputs) to double/quadruple speed.
I had a look on the source of flashrom but I am not sure if that would work with FT(2)232H.
Any comments on this?

dhendrix, 2018-02-12:
I took a quick look at the FT2232H datasheet (http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT2232H.pdf) and didn't see a way to reconfigure the IO pins to work together to transfer bits in a single direction. I think the hardware support is simply lacking for that programmer.

LMK if you spot anything that might be of use, otherwise we can close this issue.

c0d3z3r0, 2018-04-05:
Hi, I am sorry for the delay.
One question first: What is the reason for using bitbang mode instead of MPSSE SPI mode?

Since flashrom uses bitbang mode it wouldn't be that hard to add optional one/three more IO pins.
However, it seems FT(2)232H's FT1248 mode is exactly what we're looking for as it implements SPI with 1-8 bit wide data transfer.

deeplyarnab, 2018-08-13:
For FT2232H, can we implement it using Sync FIFO mode?
Page 14 & 27 Specifies Clock output Frequency in this mode as 60MHz so should work with QSPI supported chips (W25Q128FV itself supports 104MHz). Bitbang or the MPSSE protocol should be a painful option with 30Mhz or lower clocks & adaptive clock (if I'm not missing something).

c0d3z3r0, 2019-08-13:
One year is over, so it's time for a new comment :'D
Sync FIFO requires the RD and WR pins to be driven... so that is not possible.

saraazizi76, 2019-08-18:
i have a problem . i wanna use ft2232_spi programmer , but when i connect this programmer to my laptop and run this command "./flashrom -r ../output_EEPROM -p ft2232_spi" in the terminal i got this error"Unable to open FTDI device: -3 (device not found).
Error: Programmer initialization failed."

> Hi, I am sorry for the delay.
> Since flashrom uses bitbang mode I think it wouldn't be that hard to add optional one/three more IO pins. (EDIT: After reading some more in the data sheet I'm not sure about this anymore.)
> 
> However, it seems FT(2)232H's FT1248 mode is exactly what we're looking for as it implements SPI with 1-8 bit wide data transfer but I have no clue what to do with that MISO input...
> 
> Page 22 here:
> http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232H.pdf
> 
> and this one:
> http://www.ftdichip.com/Support/Documents/AppNotes/AN_167_FT1248_Parallel_Serial_Interface_Basics.pdf