flashrom/flashrom #32: Support for MB ASUS P4P800SE

VVD, 2018-01-23:
```diff -ur flashrom.orig/board_enable.c flashrom/board_enable.c
--- flashrom.orig/board_enable.c        2018-01-23 23:30:42.346381000 +0300
+++ flashrom/board_enable.c 2018-01-24 01:20:30.603030000 +0300
@@ -2360,6 +2360,7 @@
        {0x8086, 0x2570, 0x1043, 0x80f2,  0x8086, 0x24d3, 0x1043, 0x80a6, "^P4P800-E$", NULL, NULL,           P3, "ASUS",        "P4P800-E Deluxe",       0,   OK, intel_ich_gpio21_raise},
        {0x8086, 0x2570, 0x1043, 0x80a5,  0x8086, 0x24d3, 0x1043, 0x80a6, "^P4P800-VM$", NULL, NULL,          P3, "ASUS",        "P4P800-VM",             0,   OK, intel_ich_gpio21_raise},
        {0x8086, 0x2570, 0x1043, 0x80f2,  0x8086, 0x24d3, 0x1043, 0x80a6, "^P4P800-X$", NULL, NULL,           P3, "ASUS",        "P4P800-X",              0,   OK, intel_ich_gpio21_raise},
+       {0x8086, 0x2570, 0x1043, 0x80f2,  0x8086, 0x24d3,      0,      0, "^P4P800SE$", NULL, NULL,           P3, "ASUS",        "P4P800SE",              0,   OK, intel_ich_gpio21_raise},
        {0x8086, 0x2570, 0x1043, 0x80b2,  0x8086, 0x24c3, 0x1043, 0x8089, "^P4PE-X/TE$",NULL, NULL,           P3, "ASUS",        "P4PE-X/TE",             0,   NT, intel_ich_gpio21_raise},
        {0x1039, 0x0651, 0x1043, 0x8081,  0x1039, 0x0962,      0,      0, NULL,         NULL, NULL,           P3, "ASUS",        "P4SC-E",                0,   OK, it8707f_write_enable_2e},
        {0x8086, 0x2570, 0x1043, 0x80A5,  0x105A, 0x24D3, 0x1043, 0x80A6, NULL,         NULL, NULL,           P3, "ASUS",        "P4SD-LA",               0,   NT, intel_ich_gpio32_raise},
diff -ur flashrom.orig/print.c flashrom/print.c
--- flashrom.orig/print.c       2018-01-23 23:30:42.376649000 +0300
+++ flashrom/print.c        2018-01-24 00:10:02.819518000 +0300
@@ -679,6 +679,7 @@
        B("ASUS",       "P4P800-E Deluxe",      OK, "http://www.asus.com/Motherboards/Intel_Socket_478/P4P800E_Deluxe/", NULL),
        B("ASUS",       "P4P800-VM",            OK, "http://www.asus.com/Motherboards/Intel_Socket_478/P4P800VM/", NULL),
        B("ASUS",       "P4P800-X",             OK, "http://www.asus.com/Motherboards/Intel_Socket_478/P4P800X/", NULL),
+       B("ASUS",       "P4P800SE",             OK, "https://www.asus.com/supportonly/P4P800 SE/", NULL),
        B("ASUS",       "P4PE-X/TE",            NT, "http://www.asus.com/999/html/events/mb/socket478/p4pe-x-te/overview.htm", NULL),
        B("ASUS",       "P4S533-X",             OK, NULL, NULL),
        B("ASUS",       "P4S800-MX",            OK, "http://www.asus.com/Motherboards/Intel_Socket_478/P4S800MX/", NULL),
```

```
# flashrom/flashrom -p internal -w P4P800SE-ASUS-1012.001
flashrom 0.9.9-115-g25fde40-dirty on FreeBSD 11.1-RELEASE (i386)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 4, resolution: 1ns).
Found chipset "Intel ICH5/ICH5R".
Enabling flash write... OK.
Enabling full flash access for board "ASUS P4P800SE"... OK.
Found PMC flash chip "Pm49FL004" (512 kB, LPC, FWH) mapped at physical address 0xfff80000.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
```

martonmiklos, 2019-08-06:
Submitted to gerrit for review:
https://review.coreboot.org/c/flashrom/+/34764

martonmiklos, 2019-08-07:
@VVD may I ask you to post a Signed off by line either here or to gerrit in a comment (you can log in there with GH Oauth).
Once you do that I will change the signed off by line in the commit message. 

It is required by the flashrom's signed-off procedure:
https://www.flashrom.org/Development_Guidelines#Sign-off_Procedure

VVD, 2019-08-08:
You can get this patch as present for free.

martonmiklos, 2019-10-04:
@VVD, @dhendrix this patch got merged in de77ad4, so the issue could be closed.

closed 2019-11-16 by VVD