flashrom/flashrom #45: Support for partial flash write

tcs-ulli, 2018-05-25:


dhendrix, 2018-06-03:
Partial flash write is supported using layout files. See '-l' and '-i' options.

Or did you have something else in mind?

Droid-MAX, 2019-07-02:
The correspondence between the rom and the start and end addresses of the imagename in the section programming is not clearly stated in the manual.

Ed-Yang, 2022-09-29:
I tries to burn a ex. boot.bin (< 1MB> to a 128MB NOR flash.  As to my understanding, I should

1. extend boot.bin to a new bin file ex. "nor.bin" which file size is same as my NOR flash 
2. and provides a layout file to specify the region(s).

Is that possible to provide a option to just write the boot.bin to the beginning of flash, like:

flashrom ... -B boot.bin

PS.  If I am missing something, please correct me.