flashrom/flashrom #49: question: FT232RL

Zibri, 2018-06-07:
Can FT232RL be used instead of FT232H?
I see that FT232RL has bit bang mode too.
I have this breakout:
[image]
which is both 3.3v and 5v.

l29ah, 2018-08-01:
AFAIU, flashrom uses FT232H's MPSSE facility for SPI that is unavailable in FT232RL.

saraazizi76, 2019-08-18:
> AFAIU, flashrom uses FT232H's MPSSE facility for SPI that is unavailable in FT232RL.

i have another problem now. i wanna use ft2232_spi programmer , but when i connect this programmer to my laptop and run this command "./flashrom -r ../output_EEPROM -p ft2232_spi" in the terminal i got this error"Unable to open FTDI device: -3 (device not found).
Error: Programmer initialization failed."

orangecms, 2019-09-10:
Does `lsusb` list it? You may be missing the kernel module.

Check if you kernel has `USB_SERIAL_FTDI_SIO` (module `ftdi_sio`). That should give you `/dev/ttyUSB{0,1}` (two channels).

In addition to your command, you may need to supply the board type.