flashrom/flashrom #55: how to program FM25Q08 chip?

zhengsyou, 2018-08-16:
system:macOs
install mode:brew install flashrom.

i saw this chip was defined in the flashchip.h   
but flashrom can't recognize it.
why?

dhendrix, 2018-08-22:
Hi zhengsyou,
Even though the manufacturer and chip IDs have been added to the header file, there is no entry in the source file (flashchips.c).

Whoever added that chip probably did not have the ability to test it, and so support was not fully added at that time.

JakobPetersson, 2019-04-08:
@zhengsyou 
Have you seen https://github.com/flashrom/flashrom/pull/74 ?

jitendraBLE, 2020-12-24:
@JakobPetersson @dhendrix @zhengsyou Did you guys get the c file for FM25Q08 SPI flash memory?
If you know or make successful communication with this chip then please provide some reference I am struggling with reading manufacturer ID.