flashrom/flashrom #59: Account for kernel lockdown mode in error messages

pgera, 2018-08-26:
Some distros will enable the kernel lockdown mode if uefi secure boot is enabled. flashrom will error out with a generic: "ERROR: Could not get I/O privileges (Operation not permitted).
You need to be root." even if you are root. You can disable lockdown with SysRq+x.

dhendrix, 2019-08-24:
This seems doable - It seems we just need to write 'x' to /proc/sysrq-trigger?

Aside from that, is there a way of determining if kernel lockdown is currently enabled so we can try disabling it before attempting to write?

pgera, 2019-09-20:
>It seems we just need to write 'x' to /proc/sysrq-trigger?

On ubuntu at least, cat /proc/sys/kernel/sysrq returns 176, which does not allow SysRq+x by default. So the additional step of for eg. "echo "1" > /proc/sys/kernel/sysrq" is needed before SysRq+x. Overall, I think flashrom should not get into lifting lockdown. That should be left to the user and the distro. Besides, all distros may not even allow this. (That is the main point of lockdown afterall). The error message should be helpful though. 

>is there a way of determining if kernel lockdown is currently enabled 

I am not aware, but yes, this is all that should be done I think. Detect lockdown and change the error message.

guub187, 2021-08-04:
I have the same problem with latest flashrom (v1.2). That error message is shown, but it's a bit confusing. Realized that was related to kernel lockdown. The code should be changed to display a more detailed description about this.

Also, would like to use flashrom without needing to turning off Secure Boot.