flashrom/flashrom #73: ATMEL 24C01A

shellchocolat, 2019-01-28:
Hello,

I would like to dump the firmware of an Atmel 24C01A, but this flash chips is not a supported one.

I found 2 files, that could be of interest: flashchips.c and flashchips.h, but i do not know how to modify it in order to support the Atmel 24C01A chips.

If you could give me some advice about the procedure, it could be very helpful.

Thank in advance

Best regards

Hugo

martonmiklos, 2019-08-06:
The mentioned device is an EEPROM not a flash. flashrom does not supports any I2C devices at the moment. 

You might give a try to the eep24c:
http://manpages.ubuntu.com/manpages/trusty/man1/eep24c.1.html

Or if you have access to a CH341 with this utility:
https://github.com/plumbum/ch341eeprom

dhendrix, 2019-08-24:
It would be nice to add support for serial EEPROMs at some point. But there are other issues to consider, for example many don't have a way to programmatically read ID so we'd need to consider things like parsing a system devicetree (if present).

Adding Linux I2C support is easy (and actually was done in the chromiumos fork a long time ago: https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/refs/heads/master/linux_i2c.c)

closed 2019-08-24 by dhendrix