flashrom/flashrom #76: Reading old flash chip contents... Transaction error

comicsheep, 2019-03-14:
**System Information**
```

        Manufacturer: Inspur
        Product Name: SA5212M5
        Version: 00001

```
**Base Board Information**

```
        Manufacturer: Inspur
        Product Name: NF5280M5
        Version: NF5280M5
```
**BIOS Information**

```
        Vendor: Inspur
        Version: 4.0.6
        Release Date: 03/07/2019

```
**run log**
```
[root@T64 ~/rt_flash_rom]# ./flashrom -p internal -r flashrom_backup.bin
flashrom  on Linux 3.10.107-1-tlinux2_nosign_test1-0047 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel C622 Series Chipset (QS/PRQ)".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... SPI Configuration is locked down.
FREG2: Management Engine region (0x00003000-0x007dffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading flash... Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
Read operation failed!
FAILED.
[root@T64 ~/rt_flash_rom]# ./flashrom -p internal -w ./Fake_for_flash.bin
flashrom  on Linux 3.10.107-1-tlinux2_nosign_test1-0047 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel C622 Series Chipset (QS/PRQ)".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... SPI Configuration is locked down.
FREG2: Management Engine region (0x00003000-0x007dffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
FAILED.
[root@T64 ~/rt_flash_rom]#

```

comicsheep, 2019-03-14:
bios_spi_bc  register Write enable is 1 for sure

[root@T64 ~/rt_flash_rom]# setpci -s 00:1f.5 dc.b
89

CFG REG | BYTE | BIN | WE | BLE | SMM_BWP
-- | -- | -- | -- | -- | --
bios_spi_bc | 89 | 10001001 | 1 | 0 | 0

comicsheep, 2019-03-15:
-V result
```
The following protocols are supported: Parallel, LPC, FWH, Programmer-specific.
Probing for Programmer Opaque flash chip, 0 kB: Hardware sequencing reports 1 attached SPI flash chip with a density of 32768 kB.
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.

Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific).
Reading flash... Reading 33554432 bytes starting at 0x000000.
Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
HSFS: FDONE=1, FCERR=1, AEL=0, SCIP=0, PRR34_LOCKDN=0, WRSDIS=1, FDOPSS=1, FDV=1, FLOCKDN=1
HSFC: FGO=0, HSFC=0, WET=0, FDBC=63, SME=0
Read operation failed!
FAILED.
```

Th3Fanbus, 2019-03-15:
`FREG2: Management Engine region (0x00003000-0x007dffff) is locked.`

Taking this into account, flashrom failing on read/write/erase operations is expected. Why? The internal programmer cannot read or write to that region, and doing so results in an error.

It is true the error message could be a bit more descriptive, but even the non-verbose output mentions what should be done (right below the line I quoted).

comicsheep, 2019-03-15:
> `FREG2: Management Engine region (0x00003000-0x007dffff) is locked.`
>
> Taking this into account, flashrom failing on read/write/erase operations is expected. Why? The internal programmer cannot read [...]

Thanks for apply, a few more questions

For my case, which arguement should be set?
--force or --ifd -i bios image.bin

If using --ifd -i bios image.bin,  image.bin should whole flash image file or only the bios part?

comicsheep, 2019-03-18:
after some test, the following command is working, and longsoft/uefitool can parse the output file
./flashrom -p internal -r bios_back.bin --ifd -i bios

closed 2019-03-18 by comicsheep

OceanhuangCLS, 2020-02-26:
We have meet the same issue on the Intel Bakerville platform, we want to know if have any solution about this issue? Below is our run log:

[root@localhost flashrom-v1.1]# ./flashrom  -p internal -w CBR.1.00.01.bin 
flashrom v1.1 on Linux 3.10.0-957.el7.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel C620 Series Chipset Supersku".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... SPI Configuration is locked down.
FREG2: Management Engine region (0x00003000-0x00a15fff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
FAILED.
[root@localhost flashrom-v1.1]# ./flashrom  -p internal -r 111.bin --ifd -i bios
flashrom v1.1 on Linux 3.10.0-957.el7.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel C620 Series Chipset Supersku".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... SPI Configuration is locked down.
FREG2: Management Engine region (0x00003000-0x00a15fff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
Reading flash... done.