flashrom/flashrom #79: Erase fails on some W25Q128FV and W25Q128JV

CodeFetch, 2019-04-10:
My programmer is a Raspberry Pi 2.
I have two batches of W25Q128FVs and one of W25Q128JVs. 
One of the W25Q128FV batches can be erased and written perfectly using flashrom. 
The other one and the W25Q128JV batch fail to erase the chip.
All of these chips can be erased and written on OpenWrt routers.
Even stranger: The batch that can be erased using flashrom can't be erased using a TL866 II Plus, while the others work perfectly fine.

CodeFetch, 2019-04-16:
I'm running raspbian. Flashrom detects the chips correctly. It does not matter whether I pull /HOLD and /WP up or leave them floating.

Here are the outputs of one W25Q128FVSG 1210 that works and one W25Q128FVSG 1618 that does not work. The only difference between these chips is that the working one's legs are a bit more shiny.

```
pi@raspberrypi:~/artflash $ flashrom -c "W25Q128.V" -p linux_spi:dev=/dev/spidev0.0,spispeed=4000 -w outv8.img -V
flashrom p1.0-153-g0cacb11 on Linux 4.14.98-v7+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.2, GCC 6.3.0 20170516, little endian
Command line (7 args): flashrom -c W25Q128.V -p linux_spi:dev=/dev/spidev0.0,spispeed=4000 -w outv8.img -V
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing linux_spi programmer
Using device /dev/spidev0.0
Using 4000kHz clock
linux_spi_init: Using value from /sys/module/spidev/parameters/bufsiz as max buffer size.
linux_spi_init: max_kernel_buf_size: 4096
The following protocols are supported: SPI.
Probing for Winbond W25Q128.V, 16384 kB: probe_spi_rdid_generic: id1 0xef, id2 0x4018
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on linux_spi.
Chip status register is 0x00.
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
Reading old flash chip contents... done.
Erasing and writing flash chip... Trying erase function 0... 0x000000-0x000fff:EFAILED at 0x00000080! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x00000fff: 0x800
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 1... 0x000000-0x007fff:EFAILED at 0x00000080! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x00007fff: 0x4000
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 2... 0x000000-0x00ffff:EFAILED at 0x00000080! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x0000ffff: 0x8000
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 3... 0x000000-0xffffff:EFAILED at 0x00000080! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x00ffffff: 0x800000
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 4... 0x000000-0xffffff:EFAILED at 0x00000080! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x00ffffff: 0x800000
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 5... not defined. Looking for another erase function.
Trying erase function 6... not defined. Looking for another erase function.
Trying erase function 7... not defined. No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Good, writing to the flash chip apparently didn't do anything.
Please check the connections (especially those to write protection pins) between
the programmer and the flash chip. If you think the error is caused by flashrom
please report this on IRC at chat.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org, thanks!
```

And here the working one:
```
flashrom -c "W25Q128.V" -p linux_spi:dev=/dev/spidev0.0,spispeed=4000 -w outv8.img -V
flashrom p1.0-153-g0cacb11 on Linux 4.14.98-v7+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.2, GCC 6.3.0 20170516, little endian
Command line (7 args): flashrom -c W25Q128.V -p linux_spi:dev=/dev/spidev0.0,spispeed=4000 -w outv8.img -V
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing linux_spi programmer
Using device /dev/spidev0.0
Using 4000kHz clock
linux_spi_init: Using value from /sys/module/spidev/parameters/bufsiz as max buffer size.
linux_spi_init: max_kernel_buf_size: 4096
The following protocols are supported: SPI.
Probing for Winbond W25Q128.V, 16384 kB: probe_spi_rdid_generic: id1 0xef, id2 0x4018
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on linux_spi.
Chip status register is 0x00.
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
Reading old flash chip contents... done.
Erasing and writing flash chip... Trying erase function 0... 0x000000-0x000fff:EW, 0x001000-0x001fff:EW, 0x002000-0x002fff:EW, 0x003000-0x003fff:EW, 0x004000-0x004fff:EW, 0x005000-0x005fff:EW, 0x006000-0x006fff:EW, 0x007000-0x007fff:EW, 0x008000-0x008fff:EW, 0x009000-0x009fff:EW, 0x00a000-0x00afff:EW, 0x00b000-0x00bfff:EW, 0x00c000-0x00cfff:EW, 0x00d000-0x00dfff:EW, 0x00e000-0x00efff:EW, 0x00f000-0x00ffff:EW, 0x010000-0x010fff:EW, 0x011000-0x011fff:EW, 0x012000-0x012fff:S, 0x013000-0x013fff:S, 0x014000-0x014fff:S, 0x015000-0x015fff:S, 0x016000-0x016fff:S, 0x017000-0x017fff:S, 0x018000-0x018fff:S, 0x019000-0x019fff:S, 0x01a000-0x01afff:S, 0x01b000-0x01bfff:S, 0x01c000-0x01cfff:S, 0x01d000-0x01dfff:S, 0x01e000-0x01efff:S, 0x01f000-0x01ffff:W, 0x020000-0x020fff:W, 0x021000-0x021fff:W, 0x022000-0x022fff:W, 0x023000-0x023fff:W, 0x024000-0x024fff:W, 0x025000-0x025fff:W, 0x02600 [... cut ...] feafff:S, 0xfeb000-0xfebfff:S, 0xfec000-0xfecfff:S, 0xfed000-0xfedfff:S, 0xfee000-0xfeefff:S, 0xfef000-0xfeffff:S, 0xff0000-0xff0fff:S, 0xff1000-0xff1fff:W, 0xff2000-0xff2fff:S, 0xff3000-0xff3fff:S, 0xff4000-0xff4fff:S, 0xff5000-0xff5fff:S, 0xff6000-0xff6fff:S, 0xff7000-0xff7fff:S, 0xff8000-0xff8fff:S, 0xff9000-0xff9fff:S, 0xffa000-0xffafff:S, 0xffb000-0xffbfff:S, 0xffc000-0xffcfff:S, 0xffd000-0xffdfff:S, 0xffe000-0xffefff:S, 0xfff000-0xffffff:S
Erase/write done.
Verifying flash... VERIFIED.
```

CodeFetch, 2019-04-16:
I've found a very strange solution. When I write a working chip, I can write formerly non-working chips afterwards. When I reboot the Pi by cutting the power I can't write non-working new chips anymore.
After formerly non-working chips have been written once, the erase/write even works for them after a reboot and I can flash non-working new chips afterwards.
When I erase (fill with 0xff) formerly non-working chips, they can still be written after a reboot (which should not happen unless there is a changed state on the chip). 

I've confirmed this strange behavior with another Raspberry Pi 2 with another clip and cabling.

Unfortunately I don't have any new chips anymore, but I'm getting a new batch next week and some pull-up resistors (at the moment I directly connect /HOLD and /WP to VCC).

I assume this is either an electrical problem or something is wrong with the Raspberry's SPI initialization.

CodeFetch, 2019-04-23:
In my case adding a 4.7uF decoupling capacitor between VCC and Ground near the flash chip resolved the issue with the new batch. Unfortunately I can't update the [Wiki page](https://www.flashrom.org/RaspberryPi).

martonmiklos, 2019-08-07:
@CodeFetch Thanks for the tip I have extended the wiki.

CodeFetch, 2019-08-08:
Resolved

closed 2019-08-08 by CodeFetch

fips21, 2025-02-10:
> In my case adding a 4.7uF decoupling capacitor between VCC and Ground near the flash chip resolved the issue with the new batch. Unfortunately I can't update the [Wiki page](https://www.flashrom.org/RaspberryPi).

Had the same problem with arduino as serial programmer. Erase failed all the time. The capacitor solved it! Thanks a lot!

wtarreau, 2025-03-03:
Seems like it solved it as well for me on a ch341a USB programmer modded for 1.8V. Placing a 10uF capacitor between pins 1 and 8 of the I2C part (connected to GND and VCC) suffices to make the SPI properly work, and that's very easy to do.

Thank you!