flashrom/flashrom #81: Found Generic flash chip "unknown SPI chip (RDID)" (0 kB, SPI) on linux_spi.

daiaji, 2019-05-20:
Log output using the `-VVV `parameter
[flash.log](https://github.com/flashrom/flashrom/files/3197919/flash.log)
I use flashrom on the [NanoPC-T3](http://wiki.friendlyarm.com/wiki/index.php/NanoPC-T3).

saraazizi76, 2019-07-28:
sorry. i have a question.some commands are listed in man page (8) for reading, writing, ... .where this commands should be run? in terminal of ubuntu? how can i run flashrom software???

dhendrix, 2019-08-24:
The log from the OP shows a valid ID (i.e. not 0x00's or 0xff's), though the ID is unknown to flashrom. I don't think it's actually a flashchip, and without more details we can't help.

@saraazizi76 Yes, run the commands on your terminal. There are several examples in the man page (under PROGRAMMER-SPECIFIC INFORMATION), but you can also find many tutorials on-line as well.

closed 2019-08-24 by dhendrix

saraazizi76, 2019-08-25:
hi, i am using JTAG-HS2 programmer which the name of chip is FT2232C, but
when i run this command:
./flashrom -r ../output_EEPROM -p ft2232_spi

i get this error:
flashrom v1.1-rc1-68-gea0c093 on Linux 4.15.0-20-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Unable to open FTDI device: -3 (device not found).
Error: Programmer initialization failed.

i am using linux mint 19.
i don't know what is the problem

‫‪David Hendricks‬‏ <‪notifications@github.com‬‏> در تاریخ شنبه ۲۴ اوت ۲۰۱۹
ساعت ۲۲:۱۳ نوشت:‬