flashrom/flashrom #83: Status NOT WORKING for W25Q80EW chip

amirdib, 2019-05-26:
Reading Winbond W25Q80EW chip fails with the following error:
```
sudo /usr/sbin/flashrom -p ch341a_spi -r data.bin

flashrom v0.9.9-r1954 on Linux 4.15.0-45-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Winbond flash chip "unknown Winbond (ex Nexcom) SPI chip" (0 kB, SPI) on ch341a_spi.
===
This flash part has status NOT WORKING for operations: PROBE READ ERASE WRITE
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Read is not working on this chip. Aborting.
```

The W25Q80EW doesn't appear in the list of supported device eventhough it should be supported by https://github.com/flashrom/flashrom/commit/f5775448448d62c334a24618cd19692bb4a5f8a6

amirdib, 2019-05-26:
I used dated version of flashroom. If flashroom fails to recognize your ship, you should try to build it from source (new version is currently v1.1

closed 2019-05-26 by amirdib

imedhanchour, 2022-08-16:
hey 
I have the same problem , do you have the solution please  ?

cordially

vla3d, 2024-10-13:
Just meet same error. Maybe should set SPI speed. Helped to me