flashrom/flashrom #89: Flashrom error on EN25QH64 Failed to read 4100 bytes

daviem, 2019-07-08:
No matter what clip I use I get this error reading an e540 bios

Found EON flash cjhip "EN25QH64" )8192 KB, SPI) on ch341a_spi
Reading flash...
cb_in: error:LIBUSB_TRANSFER_TIMED_OUT
ch341a_spiu_spi_send_command: Failed to read 4100 bytes
Read operation failed!
FAILED

saraazizi76, 2019-07-28:
sorry. i have a question.some commands are listed in man page (8) for reading, writing, ... .where this commands should be run? in terminal of ubuntu? how can i run flashrom software???

computative, 2022-12-08:
> No matter what clip I use I get this error reading an e540 bios
> 
> Found EON flash cjhip "EN25QH64" )8192 KB, SPI) on ch341a_spi Reading flash... cb_in: error:LIBUSB_TRANSFER_TIMED_OUT ch341a_spiu_spi_send_command: Failed to read 4100 bytes Read operation failed! FAILED

How did you solve it?

krei-se, 2025-02-19:
Anyone coming across this, here's my shameful list of errors i made while getting this:

- ch341a has PIN1 not where you may expect it - check the small print for 24xx on the bottom and make sure the notch is rotated correctly. I had the whole clip setup rotated.
- take a chip you can burn (DIP8 and SOIC8 cheap on ali) and test your setup both with the chip fastened in and with a clip connected to it. Before those flash fine don't connect to something valuable ;)
- use a USB2 extension cable - this may have nothing to do with this, but i made sure anyway
- USE SHORTER WIRES FROM THE CH341A TO THE CLIP. I used a maximum of 10cm wires and since then can read and write almost anytime - i guess this might be the most important point.

Have fun.