flashrom/flashrom #95: util/getrevision.sh missing from releases

raphCode, 2019-08-23:
I discovered that the file util/getrevision.sh is missing from the release packages on github and on the coreboot website. I picked approx. 3 of the newer releases in tar.gz and zip form to try to verify this for different versions.
Without this file, no version info is compiled into the flashrom executable, running `flashrom --version` does show double empty spaces around where I would expect the version info: `flashrom  on Linux 4.4.0-53-generic (x86_64)`
Also make complains, but runs nevertheless:
```
make: ./util/getrevision.sh: Command not found
make: ./util/getrevision.sh: Command not found
Replacing all version templates with .
```

dhendrix, 2019-08-24:
Hi @raphCode,
Release tarballs actually do not use util/getrevision.sh since it relies on having the git history which we don't export as part of the release (we don't assume that the user has `git` installed on their system). Instead, there is a versioninfo.inc file included in the tarball which is generated in the _export rule in the Makefile during the release process.

Can you elaborate on the process you're using to reproduce the issue? I just checked out the latest release tarball (https://download.flashrom.org/releases/flashrom-v1.1.tar.bz2) and cloned the repo from github and ran `make`, and the results looked sane:

Using the release tarball: flashrom v1.1 on Linux 5.0.0-7-generic (ppc64le)
Using the sources from github: flashrom v1.1-rc1-69-g188127e on Linux 5.0.0-7-generic (ppc64le)

raphCode, 2019-08-24:
Hi @dhendrix,
thanks for your reply and the explanation of the release process!
I tried to reproduce the issue using a fresh copy of the release archives, but it worked correctly as you described.
I did not check the original situation since I am at different computer now, but it seems to me I got something mixed up there. Possibly I did end up having the "git Makefile" in the release directory by having it open in a text editor while swapping the directories' content...

I apologize for the false alarm and the hasty issue creation.

closed 2019-08-24 by raphCode