flashrom/flashrom #99: Tag a new flashrom release

superm1, 2019-08-29:
Any chance at tagged a new release soon?  Since https://github.com/flashrom/flashrom/commit/cb97368328bc68698ab7e58a6d692635dfb1b1c7#diff-969b60ad3d206fd45c208e266ccfed38 happened and fwupd grew flashrom support it would be good to start pushing distros to adopt it in library form so it can be used in more places.

Right now it's off by default in fwupd other than CI, but it would be good to expand. (https://github.com/fwupd/fwupd/pull/1327)

emilazy, 2019-09-03:
I'd also appreciate this, since it'd allow dropping the libftdi 0.x dependency in NixOS.

superm1, 2019-09-03:
Something else that's probably missing for a library is to have a pkg-config .pc file that can be used by projects to match to a minimum version.

superm1, 2019-11-06:
@dhendrix any chance at tagging a release soon?  All the commits are in place for fwupd to use flashrom now, we'll flip the default for fwupd to use it once you guys tag a release.

kloczek, 2020-01-31:
+1

Looks like without current flashrom is not possible to build fwupd flashrom support.

dhendrix, 2020-02-16:
Hi folks,
Flashrom-v1.2 has been tagged and released. Thanks for your patience, hopefully the next release will be easier...

Mailing list announcement: https://mail.coreboot.org/hyperkitty/list/flashrom@flashrom.org/message/C2M3PZO2EVV4UGY7YIT7CGRTPVPCWYLO/

closed 2020-02-16 by dhendrix