flashrom/flashrom #105: Meson build broken after 3d8868

asavah, 2019-10-05:
After https://github.com/flashrom/flashrom/commit/3d8868c2b46548be6885198987492d91933c9ff7 meson build is broken
```
/home/asavah/kross/host/msib450ig/lib/gcc/x86_64-msib450ig-linux-gnu/9.2.1/../../../../x86_64-msib450ig-linux-gnu/bin/ld: flashrom@sha/flashchips.c.o:(.data.rel.ro+0x34840): undefined reference to `probe_spi_st95'
/home/asavah/kross/host/msib450ig/lib/gcc/x86_64-msib450ig-linux-gnu/9.2.1/../../../../x86_64-msib450ig-linux-gnu/bin/ld: flashrom@sha/flashchips.c.o:(.data.rel.ro+0x34878): undefined reference to `spi_block_erase_emulation'
collect2: error: ld returned 1 exit status
```

Root cause:
api95.c was not added to meson.build

Suggested fix:
```
diff --git a/meson.build b/meson.build
index d778d71..0696ce4 100644
--- a/meson.build
+++ b/meson.build
@@ -321,6 +321,7 @@ srcs += 'programmer.c'
 srcs += 'sfdp.c'
 srcs += 'spi25.c'
 srcs += 'spi25_statusreg.c'
+srcs += 'spi95.c'
 srcs += 'spi.c'
 srcs += 'sst28sf040.c'
 srcs += 'sst49lfxxxc.c'
```

Question:
Are patches at least compile tested before merging?

dhendrix, 2019-10-05:
Thanks for the report, I've sent the fix upstream for review and hope to have it merged soon: https://review.coreboot.org/c/flashrom/+/35819

Patches are compile tested in the upstream code review system using jenkins, however at the moment we're only set up to run `make`. I've asked if we can have jenkins build using meson as well.

asavah, 2019-10-06:
Fixed by https://github.com/flashrom/flashrom/commit/349b5d24345c5a550c0e38cf26452c6338cc1c12
Thanks.

closed 2019-10-06 by asavah