flashrom/flashrom #106: error: redefinition of 'struct termios' on CentOS 7 ppc64le

robert-scheck, 2019-10-25:
Trying to build flashrom v1.1 on CentOS 7 for ppc64le fails like this (note that x86_64 works):

```
cc -MMD -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mcpu=power8 -mtune=power8 -I/usr/include/libusb-1.0    -D'CONFIG_DEFAULT_PROGRAMMER=PROGRAMMER_INVALID' -D'CONFIG_DEFAULT_PROGRAMMER_ARGS="''"' -D'CONFIG_SERPROG=1' -D'CONFIG_PONY_SPI=1' -D'CONFIG_BITBANG_SPI=1' -D'CONFIG_GFXNVIDIA=1' -D'CONFIG_SATASII=1' -D'CONFIG_ATAVIA=1' -D'CONFIG_IT8212=1' -D'CONFIG_FT2232_SPI=1' -D'CONFIG_USBBLASTER_SPI=1' -D'CONFIG_PICKIT2_SPI=1' -D'HAVE_FT232H=1' -I/usr/include/libftdi1 -I/usr/include/libusb-1.0   -D'CONFIG_DUMMY=1' -D'CONFIG_DRKAISER=1' -D'CONFIG_NICINTEL=1' -D'CONFIG_NICINTEL_SPI=1' -D'CONFIG_NICINTEL_EEPROM=1' -D'CONFIG_OGP_SPI=1' -D'CONFIG_BUSPIRATE_SPI=1' -D'CONFIG_DEDIPROG=1' -D'CONFIG_DEVELOPERBOX_SPI=1' -D'CONFIG_LINUX_MTD=1' -D'CONFIG_LINUX_SPI=1' -D'CONFIG_CH341A_SPI=1' -D'CONFIG_DIGILENT_SPI=1' -D'NEED_PCI=1'  -D'NEED_RAW_ACCESS=1' -D'NEED_LIBUSB0=1' -D'NEED_LIBUSB1=1' -D'HAVE_UTSNAME=1' -D'HAVE_CLOCK_GETTIME=1' -D'FLASHROM_VERSION="v1.1"' -o custom_baud.o -c custom_baud.c
In file included from custom_baud.c:23:0:
/usr/include/asm-generic/termbits.h:11:8: error: redefinition of 'struct termios'
 struct termios {
        ^
In file included from /usr/include/termios.h:40:0,
                 from /usr/include/bits/ioctl-types.h:5,
                 from /usr/include/sys/ioctl.h:29,
                 from custom_baud.c:21:
/usr/include/bits/termios.h:33:8: note: originally defined here
 struct termios {
        ^
make: *** [custom_baud.o] Error 1
```

ffontaine, 2019-11-05:
For information, we had the same issue on buildroot and we concluded that the issue was on uclibc-ng, not on flashrom, see https://patchwork.ozlabs.org/patch/1135494

robert-scheck, 2019-11-05:
CentOS 7 uses glibc, not uclibc-ng.

dhendrix, 2019-11-06:
Hi @robert-scheck
I'm able to compile flashrom at both v1.1 and the current master (b863127) on ppc64le using glibc 2.29-0ubuntu2 (according to `ldd --version`).

The error output you posted suggests that the version of glibc CentOS 7 uses is messed up, as it defines `struct termios` in both /usr/include/bits/termios.h and /usr/include/asm-generic/termbits.h.

dhendrix, 2019-11-06:
Looking at this a bit more closely, it seems a bit clearer - Linux provides asm-generic/termbits.h, while glibc provides bits/termios.h. Both define `struct termios`.

Looking thru the link @ffontaine provided, Thomas Petazzoni did some nice sleuthing and found:

> Then, if you look at glibc, indeed until commit
> d4795e4a43e6f0c221bc5dc64c612206a21a177b, the powerpc-specific
> ioctl-types.h was just including <termios.h>, but in this commit, they
> changed it back to define the different macros and structures.

robert-scheck, 2019-11-06:
Has the patch mentioned by @ffontaine a chance to make it upstream? Or is upstream's recommendation to keep this patch downstream only? The patch solves the build failure and doesn't at least introduce any new build failures on Fedora 29-32, EPEL 7/8 (all supported archs per release).

dhendrix, 2019-11-07:
@robert-scheck Yes, it's in upstream glibc-2.20 and newer. Using docker, I see that CentOS 7 uses glibc 2.17:

```
$ docker pull centos:centos7
$ docker run centos:centos7 ldd --version
ldd (GNU libc) 2.17
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

There doesn't appear to be a CentOS 8 or Stream release for ppc64le (docker or otherwise). But as you point out, Fedora should be OK (I tried with F26, but didn't go back further than that).

robert-scheck, 2019-11-07:
@dhendrix, so this means EPEL 7 should keep the patch mentioned by @ffontaine downstream only?

dhendrix, 2019-11-07:
@robert-scheck Good question... In general I think the answer is yes, patches like this should probably be kept downstream. However it's a pretty small patch in this case and perhaps we can accommodate it upstream by adding glibc version macros to apply the hack only if needed.

I'll try to take a look at it tomorrow or this weekend when I have a bit more time.

robert-scheck, 2020-02-11:
@dhendrix, any chance for a decision?