flashrom/flashrom #111: Does flashrom supports serial nand chips?

AsmitaJha08, 2019-11-26:
I found that winbond flash chips supported by flashrom are all nor based. Can I add nand based serial flash chips?

Thanks & Regards
Asmita

dhendrix, 2019-11-26:
What OS are you primarily targeting? For Linux, this usually goes thru the MTD subsystem which we support (see: linux_mtd.c).

AsmitaJha08, 2019-11-27:
I am running flashrom on my Linux system and my serial nand flash chip is connected using ft2232.
And I am running flashrom command but flash chip not detected.

dhendrix, 2019-11-27:
Thanks for the info. Unfortunately flashrom does not support directly writing to NAND flash chips... yet. I'll look into it more this weekend. I think SPI NAND flash is going to increase in popularity, so it will be a nice feature to have.

minhng99, 2019-12-22:
Same here, I was trying to read the MX35LF1GE4AB but flashrom can't detect it, I thought my CH341 was faulty but I also tried with FT232 but getting the same result, here is the datasheet of the chip I was going to read:

https://www.macronix.com/Lists/Datasheet/Attachments/7666/MX35LF2GE4AB,%203V,%202Gb,%20v1.7.pdf

meanwhile... does anyone know how to read/write into this chip?

AsmitaJha08, 2020-01-24:
@dhendrix , were you able to able to find some way for SPI NAND flash. I want to contribute in its development, kindly suggest in what way should I proceed for SPI NAND flash?

Thanks & Regards
Asmita

dhendrix, 2020-02-15:
@AsmitaJha08 Unfortunately I haven't had much time to look into this. Others have asked about it, so if you're feeling motivated then it would be great to see some patches and get others involved :-)

The main issue seems to be that NAND flash chips often require logic to mark bad blocks. I'm unaware of any standards here, as far as I know it's implementation-defined. Basically, flashrom does not know how the user's OS will track bad blocks, and the two must agree or data loss is pretty much guaranteed.

Some NAND flash chips might have logic to remap bad blocks on their own, without the user having to keep track of them. That would be an easy target to start with since we'd only need to worry about adding the read/program command sequences.

lynxis, 2022-09-15:
It might be worth looking at the linux soft ecc.

lap012, 2024-03-08:
> Same here, I was trying to read the MX35LF1GE4AB but flashrom can't detect it, I thought my CH341 was faulty but I also tried with FT232 but getting the same result, here is the datasheet of the [...]

Have you found a solution for this ?

I'm trying to dump this same chip (MX35LF1GE4AB) and stuck at flashrom does not support NAND yet.

Is there any tool (Linux or Windows) to dump NAND flashes ?

minhng99, 2024-03-08:
> Have you found a solution for this ?
>
> I'm trying to dump this same chip (MX35LF1GE4AB) and stuck at flashrom does not support NAND yet.
>
> Is there any tool (Linux or Windows) to dump NAND flashes ?

Use SNANDer