flashrom/flashrom #112: Does it have support for tl866cs/tl866a

Lucas-bash, 2019-11-26:
I want to know I have one, and I use ubuntu. I don't wanna change my OS.

dhendrix, 2019-11-27:
No, not yet anyway. Is there a protocol spec or some existing open-source code we can use as a reference?

Lucas-bash, 2019-11-30:
hi howdy I sow some files in github, those were for running it under wine
app.

https://github.com/wholder/MiniPro

https://github.com/jfcandidofilho/minipro-aur

please see this both links I've send you.

thx for you feedback

Lucas-bash, 2019-11-30:
https://github.com/search?q=tl866

https://github.com/ProgHQ/open-tl866

https://github.com/radiomanV/TL866

https://github.com/gaggi/27c160-tl866-adapter

https://github.com/mafe72/27c160-tl866-adapter

https://github.com/wholder/MiniPro

https://github.com/jfcandidofilho/minipro-aur

closed 2022-05-10 by Th3Fanbus