flashrom/flashrom #120: GigaDevice chips GD25B64CSIG & GD25B128CSIG

rivehafintosh, 2020-01-19:
Unless I'm mistaken.... It seems to be lacking support entirely for the smaller chip.
And perhaps it's just me but the larger chip although being detected as an identified chip, regardless of what chip type is selected manually I cannot get a consistent read. I'm unsure if it's because the chip isn't an exact match (which seems likely) or if it's due to something else. I'm using an RPi3B+ and a test  clip. I know my power supply is adequate for my Pi, jumper wires are brand new, and short...I have tried multiple spispeed= parameters to no avail. Interested in figuring this out be it helping someone test things with my chips, or attempting to edit my own drivers from the datasheet.. but I will admit I would be completely out of my depth when it comes to code, so here's hoping the guides in the Wiki are REALLY good.

iddq, 2020-01-21:
do you have arduino due?

dhendrix, 2020-01-21:
This can happen if some pins are left floating. /HOLD and /WP need to be connected to VCC 3.3V, even if you don't intend to use them.

The wiki (https://www.flashrom.org/RaspberryPi) has another tip that you might need to try: _If your flash chip is detected but your read/write/verify operations tend to fail, try to add decoupling capacitors (one 100nF and one 4.7uF ceramic capacitor is preferred) close to the flash chip's power pin._

rivehafintosh, 2020-01-21:
I actually just found the opposite to be the case, at least this holds true for the GD25B128CSIG that I was having troubles with... if I understand things correctly.. now that I removed VCC 3.3V from 2 pins I'm getting a matching checksum on my read dumps using the -c "GD25B128B/GD25Q128B" with an spispeed=1000 - @iddq  I do not have an arduino duo, but I have an Uno R3, but no level shifting boards to make use of it. Thus the RPi3B+ has been my default. I'm about to try the GD25B64C chip next and see if anything changes now that I have things wired a little differently. Regardless @dhendrix thanks for your swift response and the tidbit from the flashrom.org/RaspberryPi section.. Hopefully I won't need to end up using the extra capacitors but I appreciate the advice/direction.. if it comes to it now I know what I'll be trying next...

dhendrix, 2020-01-21:
Interesting! Well, glad it's working out for you. I'm going to close this issue for now, but please feel free to re-open or comment if you're still having problems or find anything interesting - perhaps we can improve the wiki.

rivehafintosh, 2020-01-21:
After testing the smaller chip again it seems I can't even detect a generic chip. Something of note though, some of the LEDs on the board the chip is soldered to light up and flash with the clip connected, but still no dice.

dhendrix, 2020-01-21:
Ah yes, in-system programming (ISP) is tricky since you are backpowering other components on the board (as you say, LEDs light up). There are some suggestions on the wiki for dealing with that here https://flashrom.org/ISP

Long story short is that the RPi is probably not sufficient to power on the NOR flash with other stuff on the circuit, and you also need to be careful about other SPI masters powering up (e.g. embedded controller, management engine, PSP, etc).

dhendrix, 2020-01-23:
Re-opening this. I'm curious what solution you come up with and if it's something we should add to the wiki!

iddq, 2020-01-23:
Is it a desoldered chip?

rivehafintosh, 2020-01-26:
So just a little update, and correction... I was able to successfully read and write to an in-circuit chip (GD25B127CSIG) using my Raspberry Pi 3B+ and a Pomona test clip. Utilizing 6 out of 8 pins and the following line: sudo flashrom -VVV -p linux_spi:dev=/dev/spidev0.0,spispeed=1000 -c "GD25Q127C/GD25Q128C"
Notes: 
Corrupt reads with spispeed higher than 1000
Corrupt reads with VCC 3.3v connected to HOLD/WP
This process for these chips was because I needed to clear a password on an HP Folio 1040 G3 and I was able to successfully do so by clearing the VSS Store from my dump and replacing the VSS Store volume with an empty VSS Store volume from the most .bin file available from HP. Using UEFITool_NE to find the correct store it's GUID and extract the matching volume and UEFITool 27 to replace the volume after locating correct GUID was all that was required for clearing the password without losing things like MAC, UUID, Feature-byte, etc..
Seeing as the GD25B128CSIG was the only chip needing interaction to solve my problem I will probably not spend any further time troubleshooting connectivity with the GD25B64CSIG chip also found on the board as it just isn't necessary. If anyone needs/wants more information or help, be it testing or otherwise with anything at all please don't hesitate to ask. I'm more than willing to play an active role in helping the flashrom development team/community benefit in whatever way I can offer. Though I suspect what I can do is very limited as I'm simply a technician not a developer myself. 
@dhendrix if you want me to write something up for the wiki related to my experience/procedure I am more than willing to put something together. (Already have a fritzing file with wiring, and scripts for most of the process. wouldn't mind sharing them simply just lack my own platform to share such things) Sorry for the delay in a response, I was rather busy the last few days. Anyways, looking forward to hearing from you & let me offer a big thanks for everyone who replied with helpful information, and all the developers who make flashrom available.. Really is a wonderful tool, keep up the awesome work!