flashrom/flashrom #121: Please help add support for S25FS512S

GarthKiepper, 2020-01-27:
Hello,

Please add support for S25FS512S. The datasheet is located here: https://www.cypress.com/file/216376/download

I am trying to dump the SPI flash for the Tesla Model 3 autopilot computer. This is part of an automotive security research project I am doing.

I ran the following command:
flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=1024 -VVV

Here is the output, plus my wiring diagram:
https://docs.google.com/document/d/1XvWd8BaR-gnYAS-H_iBbYfKjL1RfijFcAdcFkTEuoiQ/edit

Please let me know if there are any further commands I should run.

iddq, 2020-01-27:
you can try with -c "MX66L51235F/MX25L51245G"

lewurm, 2020-01-27:
Here a patch that allowed us to _read_ the SPI flash with the setup described by @GarthKiepper:
https://github.com/lewurm/flashrom/commit/9764820296b19004652b485a06908257aa33bac8

Erase doesn't work (see error in commit message). Thus, writing is also messed up as it seems to depend on the erase command 😕

lewurm, 2020-01-27:
Nevermind, I think I got it working: https://review.coreboot.org/c/flashrom/+/38596

dhendrix, 2020-01-28:
@lewurm Thanks, that works pretty well for me if I change FEATURE_4BA_EAR7 to FEATURE_4BA_NATIVE and only use the DC erase opcode. I'm testing with a Raspberry Pi 3B 1.2 with spispeed=8000.

One thing I noticed in @GarthKiepper's diagram is that he's using a level shifter, and those can be awfully slow, like hundreds of KHz. So you might try running at a very slow speed to test. When doing so, try using a layout file so you're only operating on part of the ROM, for example:
echo '1100000:11fffff r1' > layout.txt
flashrom -V -l layout.txt -i r1 ...

lewurm, 2020-01-28:
Thanks for your comment @dhendrix :)

> only use the DC erase opcode.

So you are suggesting to remove `d8`, `60` and `c7`? I've tested it and it looks good. Also `s/FEATURE_4BA_EAR7/FEATURE_4BA_NATIVE/` seems to work. I've updated the patch set.

dhendrix, 2020-01-30:
Sorry, I should have been more clear. I suggest using dc since it uses a 4-byte address, rather than `d8` which uses a 3-byte address in the instruction + 1 byte in the extended address register. I'm not entirely sure why `d8` didn't work for me, but we can investigate that whenever someone has time.

Bulk erase instructions (`60` and `c7`) should both work fine since they don't require an address at all, and in fact I tested both of those instructions and they worked as expected.