flashrom/flashrom #125: flashrom 1.2 doesn't build on aarch64-linux

mmahut, 2020-02-12:
The flashrom 1.2 fails to build on ```aarch64-linux``` platform with following, example on NixOS:

```warning: ignoring the user-specified setting 'restrict-eval', because it is a restricted setting and you are not a trusted user
warning: ignoring the user-specified setting 'show-trace', because it is a restricted setting and you are not a trusted user
these derivations will be built:
  /nix/store/xn3j14fkrwx7kp5hw8y6w7q70lw3bqh9-flashrom-v1.2.tar.bz2.drv
  /nix/store/i0jr9ac4rhdbc5c0slypi35airr2sn8p-flashrom-1.2.drv
these paths will be fetched (0.34 MiB download, 1.69 MiB unpacked):
  /nix/store/01gfnyhy4bhzkpmbwmjxnwwr3j3ygxdn-pciutils-3.6.4
  /nix/store/8slc61jhfjjc72lvrllibq63sir8wwzk-libftdi-0.20
  /nix/store/9f4b297y86nhs23xwvbcb0fgxsgcwz77-libusb-compat-0.1.5
  /nix/store/l8a0sb0af1nsf93syrpylcmd6i0va2sk-libusb-1.0.23-dev
  /nix/store/mchviyg4qs0na767p7i0qh9bbrhqdyaj-libusb-compat-0.1.5-dev
copying path '/nix/store/01gfnyhy4bhzkpmbwmjxnwwr3j3ygxdn-pciutils-3.6.4' from 'https://cache.nixos.org'...
copying path '/nix/store/9f4b297y86nhs23xwvbcb0fgxsgcwz77-libusb-compat-0.1.5' from 'https://cache.nixos.org'...
building '/nix/store/xn3j14fkrwx7kp5hw8y6w7q70lw3bqh9-flashrom-v1.2.tar.bz2.drv'...
copying path '/nix/store/l8a0sb0af1nsf93syrpylcmd6i0va2sk-libusb-1.0.23-dev' from 'https://cache.nixos.org'...
copying path '/nix/store/mchviyg4qs0na767p7i0qh9bbrhqdyaj-libusb-compat-0.1.5-dev' from 'https://cache.nixos.org'...
copying path '/nix/store/8slc61jhfjjc72lvrllibq63sir8wwzk-libftdi-0.20' from 'https://cache.nixos.org'...
trying https://download.flashrom.org/releases/flashrom-v1.2.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  358k  100  358k    0     0   509k      0 --:--:-- --:--:-- --:--:--  509k
building '/nix/store/i0jr9ac4rhdbc5c0slypi35airr2sn8p-flashrom-1.2.drv'...
unpacking sources
unpacking source archive /nix/store/b4l06sy4cmdx0g673bfhjv25rvr5iwdk-flashrom-v1.2.tar.bz2
source root is flashrom-v1.2
setting SOURCE_DATE_EPOCH to timestamp 1581291272 of file flashrom-v1.2/versioninfo.inc
patching sources
updateAutotoolsGnuConfigScriptsPhase
configuring
no configure script, doing nothing
building
build flags: SHELL=/nix/store/2vbjxjz6yxdl74fcq2dhk45w07gk52dr-bash-4.4-p23/bin/bash
Replacing all version templates with v1.2.
Checking for a C compiler... found.
Target arch is unknown. Aborting.
make: *** [Makefile:1182: compiler] Error 1
builder for '/nix/store/i0jr9ac4rhdbc5c0slypi35airr2sn8p-flashrom-1.2.drv' failed with exit code 2
error: build of '/nix/store/i0jr9ac4rhdbc5c0slypi35airr2sn8p-flashrom-1.2.drv' failed```

dhendrix, 2020-02-13:
Hello Marek,
Thanks for the report. For what it's worth aarch64 compilation generally works (for example Fedora: https://koji.fedoraproject.org/koji/buildinfo?buildID=1458754).

The issue seems to be with the architecture detection part of the Makefile. I'll take a look into it shortly.

dhendrix, 2020-02-15:
@mmahut  - The issue seems to be that macro that indicates target architecture, such as __arm__ or __aarch64__, is not being defined: https://github.com/flashrom/flashrom/blob/master/platform.h#L52

I'd be surprised if the problem you are experiences is only happening in flashrom. Do you happen to know of other programs that have similar checks for target architecture?

Also, can you run `gcc -E archtest.c` within the flashrom directory and paste the result somewhere? The last line should be a valid architecture. Here is what it looks like on my Raspberry Pi running Debian: https://gist.github.com/dhendrix/3e80d84898ccb57bda29b016eac43d41

Is there an easy way to bootstap a NixOS environment in qemu using a non-NixOS host? I'm trying to follow the docs on nixos.wiki, but much of it seems to assume a NixOS host.

jtojnar, 2020-02-18:
With meson (https://github.com/NixOS/nixpkgs/pull/80293) the failure is different:

```
The Meson build system
Version: 0.52.1
Source dir: /build/flashrom-v1.2
Build dir: /build/flashrom-v1.2/build
Build type: native build
Project name: flashromutils
Project version: v1.2
C compiler for the host machine: /nix/store/ajd5q34w4gfr43r4ibcc712rqdw20d7s-gcc-wrapper-9.2.0/bin/cc (gcc 9.2.0 "gcc (GCC) 9.2.0")
C linker for the host machine: GNU ld.bfd 2.31.1
Host machine cpu family: aarch64
Host machine cpu: aarch64
Compiler for C supports arguments -Wwrite-strings: YES
Compiler for C supports arguments -Wno-unused-parameter -Wunused-parameter: YES
Compiler for C supports arguments -Wno-address-of-packed-member -Waddress-of-packed-member: YES
Compiler for C supports arguments -Wno-enum-conversion -Wenum-conversion: NO
Compiler for C supports arguments -Wno-missing-braces -Wmissing-braces: YES
Checking for function "clock_gettime" : YES
Checking for function "strnlen" : YES
Check usable header "sys/utsname.h" : YES
Found pkg-config: /nix/store/16c33962b8drgqi9ffzg2zbmnch0yr7q-pkg-config-0.29.2/bin/pkg-config (0.29.2)
Run-time dependency libusb-1.0 found: YES 1.0.23
Run-time dependency libpci found: YES 3.6.4
Run-time dependency libftdi1 found: YES 1.4
Build targets in project: 3
Found ninja-1.10.0 at /nix/store/lfy35ap17qb22vkcdj74903ycbc7yfgy-ninja-1.10.0/bin/ninja
meson: enabled parallel building
building
build flags: -j64 -l64
[1/140] Compiling C object 'flashrom@sha/nicrealtek.c.o'.'.K
FAILED: flashrom@sha/nicrealtek.c.o
/nix/store/ajd5q34w4gfr43r4ibcc712rqdw20d7s-gcc-wrapper-9.2.0/bin/cc -Iflashrom@sha -I. -I.. -I/nix/store/xl9rl2jglsf3pi8nyw6lpk9iyf982c2s-libusb-1.0.23-dev/include/libusb-1.0 -I/nix/store/f2idqi2s1lpkmlx133zbxly0spm0dac8-pciutils-3.6.4/include -I/nix/store/xki3sydlq6pkzmbg04xcslk3ndag6vn5-libftdi1-1.4/include/libftdi1 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ 'flashrom@sha/nicrealtek.c.o' -MF 'flashrom@sha/nicrealtek.c.o.d' -o 'flashrom@sha/nicrealtek.c.o' -c ../nicrealtek.c
../nicrealtek.c:133:2: error: #error PCI port I/O access is not supported on this architecture yet.
  133 | #error PCI port I/O access is not supported on this architecture yet.
      |  ^~~~~
[2/140] Compiling C object 'flashrom@sha/satamv.c.o'.
FAILED: flashrom@sha/satamv.c.o
/nix/store/ajd5q34w4gfr43r4ibcc712rqdw20d7s-gcc-wrapper-9.2.0/bin/cc -Iflashrom@sha -I. -I.. -I/nix/store/xl9rl2jglsf3pi8nyw6lpk9iyf982c2s-libusb-1.0.23-dev/include/libusb-1.0 -I/nix/store/f2idqi2s1lpkmlx133zbxly0spm0dac8-pciutils-3.6.4/include -I/nix/store/xki3sydlq6pkzmbg04xcslk3ndag6vn5-libftdi1-1.4/include/libftdi1 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ 'flashrom@sha/satamv.c.o' -MF 'flashrom@sha/satamv.c.o.d' -o 'flashrom@sha/satamv.c.o' -c ../satamv.c
../satamv.c:192:2: error: #error PCI port I/O access is not supported on this architecture yet.
  192 | #error PCI port I/O access is not supported on this architecture yet.
      |  ^~~~~
[3/140] Compiling C object 'flashrom@sha/rayer_spi.c.o'.
FAILED: flashrom@sha/rayer_spi.c.o
/nix/store/ajd5q34w4gfr43r4ibcc712rqdw20d7s-gcc-wrapper-9.2.0/bin/cc -Iflashrom@sha -I. -I.. -I/nix/store/xl9rl2jglsf3pi8nyw6lpk9iyf982c2s-libusb-1.0.23-dev/include/libusb-1.0 -I/nix/store/f2idqi2s1lpkmlx133zbxly0spm0dac8-pciutils-3.6.4/include -I/nix/store/xki3sydlq6pkzmbg04xcslk3ndag6vn5-libftdi1-1.4/include/libftdi1 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION=" [... cut ...]
# 1 "/nix/store/cvxy9dsgx9fbfq4hgkgi4faj25lb7dwa-glibc-2.30-dev/include/asm-generic/param.h" 1 3 4
# 23 "/nix/store/cvxy9dsgx9fbfq4hgkgi4faj25lb7dwa-glibc-2.30-dev/include/asm/param.h" 2 3 4
# 6 "/nix/store/cvxy9dsgx9fbfq4hgkgi4faj25lb7dwa-glibc-2.30-dev/include/linux/param.h" 2 3 4
# 29 "/nix/store/cvxy9dsgx9fbfq4hgkgi4faj25lb7dwa-glibc-2.30-dev/include/bits/param.h" 2 3 4
# 32 "/nix/store/cvxy9dsgx9fbfq4hgkgi4faj25lb7dwa-glibc-2.30-dev/include/sys/param.h" 2 3 4
# 143 "platform.h" 2
# 2 "archtest.c" 2

# 2 "archtest.c"
"arm"
```

golddranks, 2020-07-06:
I'm being affected by the same issue. Here's some additional notes:

- Trying to build on NixOS, AArch64, on Raspberry Pi 4, in order to get the Raspberry Pi EEPROM update script working.
- I tried an earlier version of flashrom, notably 0.9.7 which seems to be the version the official Raspberry Pi OS packages ( https://github.com/raspberrypi/noobs/blob/master/buildroot/package/flashrom/flashrom.mk ), but it fails also. ( https://github.com/NixOS/nixpkgs/blob/ea49ac04966380be9bc77c62c0d01862b515e6a6/pkgs/tools/misc/flashrom/default.nix )
- The reason why the official Raspberry Pi OS is able to provide a flashrom build is most likely that they don't even target AArch64 but the 32-bit arch, not hitting this problem.

The build system of flashrom doesn't seem familiar to me – is the Makefile hand-crafted or is it generated by something?

golddranks, 2020-07-06:
So, some more notes. First of all,

> The issue seems to be that a compiler macro that indicates target architecture, such as __arm__ or __aarch64__, is not being defined: https://github.com/flashrom/flashrom/blob/master/platform.h#L52

This doesn't seem to be true, the macro is defined at least on my system.

Tried building 1.2 with the Makefile (not using meson). It fails here: https://github.com/flashrom/flashrom/blob/1.2.x/Makefile#L1185
... and the reason it fails is because the make variable `ARCH` contains unexpected contents. I edited the Makefile to print it, it's:

```
bit outside of fd_set selected arm
```

What that means, is beyond me.

golddranks, 2020-07-06:
An explanation for this value. From here:

```override ARCH := $(strip $(call debug_shell,$(CC) $(CPPFLAGS) -E archtest.c 2>/dev/null | grep -v '^\#' | grep '"' | cut -f 2 -d'"'))```

Take `cat archtest.log | grep -v '^\#' | grep '"' | cut -f 2 -d'"'`, where `archtest.log` is the log pasted earlier in this thread. The result is:

```
bit outside of fd_set selected
arm

```

jtojnar, 2020-07-06:
Looks like some compiler warning. What do you get if you do not cut it?

golddranks, 2020-07-06:
I now have a patch that fixes the problem. I'm trying to figure out how to submit it currently, but the instructions here don't work: https://www.flashrom.org/Development_Guidelines#Using_Github

golddranks, 2020-07-06:
Managed to submit it! https://review.coreboot.org/c/flashrom/+/43140 The wiki seemed to have the repo address a bit wrong; not: `git remote add -f upstream ssh://<gerrit_username>@review.coreboot.org:29418/flashrom.git.` but `git remote add -f upstream ssh://golddranks@review.coreboot.org:29418/flashrom`.

golddranks, 2020-07-06:
@jtojnar The output parsing script at first strips the lines starting with `#`, which seem to be preprocessor "stack" annotations; however, after that, it just finds lines with `"` which seems dangerously non-robust parsing method. Here, this line also happens to match: `  __attribute__((__warning__ ("bit outside of fd_set selected")));`, which explains the error.

To me, this seems just an oversight in the output parsing, and the built binary seems to function fine with my patch.

dhendrix, 2020-07-19:
> Managed to submit it! https://review.coreboot.org/c/flashrom/+/43140 The wiki seemed to have the repo address a bit wrong; not: `git remote add -f upstream ssh://<gerrit_username>@review.coreboot.org:29418/flashrom.git.` but `git remote add -f upstream ssh://<gerrit_username>@review.coreboot.org:29418/flashrom`.

Thanks for pointing that out, I've fixed the wiki (hopefully).

golddranks, 2020-07-20:
Thanks! Btw. about the patch, I'm going to fix it during the following days, hopefully it sticks.

golddranks, 2020-08-24:
https://review.coreboot.org/c/flashrom/+/43140 Sorry, no luck yet. It's been reviewed a long time ago, but nobody seems to go and merge it...

golddranks, 2020-08-29:
My PR was merged, so this will be fixed eventually. No idea when the next release is, though.