flashrom/flashrom #127: GigaDevice 25B256D support

aziztcf, 2020-02-16:
Hey, I'm trying to look through the datasheet to spot the differences between 25B/25Q series so I could maybe try and see if I can bodge a patch for flashrom. Any help would be greatly appreciated!

[GD25**B**256D datasheet](https://www.gigadevice.com/datasheet/gd25b256d/)

[GD25**Q**256D datasheet](https://www.gigadevice.com/datasheet/gd25q256d/)

dhendrix, 2020-02-19:
Hi @aziztcf ,
Like you say, the B model is "QE=1 permanently" according to the ordering information in the datasheet, so it makes sense for it to only support IO functions on certain pins that have other uses in non-QE modes.

The ID bytes are the same on both chips. For Flashrom's purposes they're the same, but obviously we can't clear QE mode from the B model.

NudistBeaaach, 2025-09-06:
> Hi! Has any progress been made with this chip? I've got an HP ProDesk 600 G6 with the GD25B256D I need to remove the password from and I can't get it to work with the stable or git versions of flashrom

Hi did you have any news on this ? Does it means that the `GD25B256D` can only be read by QUAD-SPI ? And is flashrom able to use QUAD-SPI ?

yonatan8070, 2025-09-06:
I didn't end up fixing the computer (I did manage to completely brick it though...).
Though I did manage to read the chip, it worked once I switched from using an Arduino + level shifter (if memory serves me right) to directly using the GPIO on a Raspberry Pi 1 (just what I had lying around).
Hope this helps