flashrom/flashrom #130: 775Twins-HDTV - Add Support - All Logs included

direstraits96, 2020-03-02:
Manufacturer: ASRock
Model: 775Twins-HDTV
Notes: Can read, but can't flash.

I can test everything with this board. I have spare eeproms and a bios
backup. I'm trying to flash by hotswaping the chip.

I think i need to enable "w83627ehf_gpio22_raise_2e" in
"boards_enable.c" I tried, but flashrom can't detect the board. I had
the correct PCI ID's.

[flashrom_log.txt](https://github.com/flashrom/flashrom/files/4272939/flashrom_log.txt)
[lspci_log.txt](https://github.com/flashrom/flashrom/files/4272940/lspci_log.txt)
[superiotool_log.txt](https://github.com/flashrom/flashrom/files/4272941/superiotool_log.txt)

Th3Fanbus, 2020-07-26:
Without a log that shows the write failure, it's impossible to know what went wrong.

direstraits96, 2020-07-27:
Here are full logs. But now I have to run with ":laptop=this_is_not_a_laptop" but it's a desktop board. I'm using the latest version from github compiled by me. I can read the chip fine. But I can't write or erase.

[erase_full.txt](https://github.com/flashrom/flashrom/files/4979125/erase_full.txt)
[full_no_op.txt](https://github.com/flashrom/flashrom/files/4979126/full_no_op.txt)
[write_full.txt](https://github.com/flashrom/flashrom/files/4979127/write_full.txt)