flashrom/flashrom #131: Winbond W25Q64.Q Support

stonepresto, 2020-03-05:
Flashrom is unable to recognize the flash chip. Any plans for support of the W25Q64.Q series chips?

dhendrix, 2020-03-05:
I suspect you are referring to the quad enabled variants, which AFAIK don't have a unique device ID. They are (AFAIK) the same as the W25Q64.V and W25Q64.W versions, only with QE=1.

If that's the case, then what really matters is whether or not your SPI controller supports them. From flashrom's perspective they are the same as other parts, it's really just a matter of whether or not your controller is set up for quad SPI.

stonepresto, 2020-03-06:
Yeah after some further investigation I do not believe my controller supports it. Unfortunately I will need to spend a lot *more* time investigating exactly what the issue is.

closed 2020-03-06 by stonepresto