flashrom/flashrom #138: Question: Reading complete Chip, modify some content & write it back.

DocSneider, 2020-04-04:
Hello I have to "patch" a bunch of routers, to get OpenWrt running on them.
Therefore I need to dump the flash, make some changes on it and write back the "patched" image.

I make 2 dumps and compare them checksums to get some safety.
Is this unnecessary or a good way to go? I could safe some time by skipping the second dump.
```
#compare dumps
	dump_hash1=$(md5sum "${file}Nr01.bin" | cut -f1 -d" ") # only get the hash value
	dump_hash2=$(md5sum "${file}Nr02.bin" | cut -f1 -d" ")
	if [ "${dump_hash1}" = "${dump_hash2}" ]; then
		printf "hashs are identical - continue...\n"
	else
		printf "ERROR: dumping flash content --> md5sum´s doesn´t match - please check connection & retry.\n"
		exit_error
	fi
```
After patching, I write the file back and flashrom does read the whole flash again.
**Is there a way to commit the dumps taken right before to flashrom, instead of reading the flash again?**
It would be nice because I could save some time :-)

dhendrix, 2020-04-04:
Great question - You do not need to read the chip twice if you are certain that the flash chip will not be accessed by other software or hardware while flashrom is running.

The reason most people read the flash chip twice before attempting to write is to gain confidence that nothing else in the system will interfere with the write operation:
- If running flashrom inband from the device's OS, it might not have exclusive access to the chip and other masters may attempt to read/write, thus altering the chip's address counter.

- If running flashrom from another system, for example when using an external programmer such as CH341 or Dediprog, it may result in powering on other components which may attempt to read the flash chip and result in inconsistent results.

> **Is there a way to commit the dumps taken right before to flashrom, instead of reading the flash again?**

This feature was considered and implemented in some forks of flashrom such as ChromiumOS's version (the --diff-file option), however it was deemed too confusing and risky to be included in upstream flashrom.

For example, some programmers such as ECs in laptops may need to be paused so that they don't write data to the flash during the firmware update process. So in that case it's important for flashrom to pause the EC, read, write, and resume the EC. In that sort of situation feeding a dump into flashrom could easily lead to inconsistent state and brick the system.

DocSneider, 2020-04-05:
Thanks a lot for your comprehensive answer :-)

Because I am using a CH341 and I am reading the flash "in system" I will go on with reading the flash two times.

> 
This feature was considered and implemented in some forks of flashrom such as ChromiumOS's version (the --diff-file option), however it was deemed too confusing and risky to be included in upstream flashrom.

Ok that makes sense.

closed 2020-04-05 by DocSneider