flashrom/flashrom #139: Stuck at "Reading flash... " by using flashrom on Debian's VM

Heexos, 2020-04-11:
Hi,
I start a project where I have to reprogram a flash. I am using a CH341A programmer.
I need to run some Bash scripts so I am using my Debian's VM for my project. My VM recognize well the Programmer:
```
userir@debian:~ $ lsusb
Bus 001 Device 004: ID 1a86:5512 QinHeng Electronics CH341 in EPP/MEM/I2C mode, EPP/I2C adapter
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
but when I try to create a flash backup, it remains stucked at "Reading flash..."
```
userir@debian:~ 1 $ sudo flashrom --programmer ch341a_spi -r backup.bin
flashrom  on Linux 4.19.0-6-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Eon flash chip "EN25QH128" (16384 kB, SPI) on ch341a_spi.
===
This flash part has status UNTESTED for operations: PROBE READ ERASE WRITE
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading flash... 
```
I don't think that my problem comes from my Programmer connections becasue with a ch341a usb programming tool for Windows, I am able to read and program my flash.

Does anyone have any solutions?

PS: I'm a newbie in that domain so all of that stuff remains unclear for me, so if you have any question, don't hesitate!

Th3Fanbus, 2020-07-26:
How long did you wait for? The CH341A is *slow*, and since your chip is big, it may take several minutes. If you want to see progress, you can use the `-VVV` command-line switch.

DerEchteJoghurt, 2024-08-06:
On later versions progress is available with the `--progress` flag. 

(For everyone else ending up here from a search, wondering why `-VVV` does not show any progress information).

tlaurion, 2024-08-07:
Yeah, waiting for --progress to come back here too to remove debug output parsing workaround d under Heads which consumes a lot of CPU and time, and is also bogus at times.

Why progress is not brought back @Th3Fanbus ?

Th3Fanbus, 2024-08-07:
@tlaurion Um, this is a 4-year-old issue that does not seem to be related to the issue you have. Moreover, I have not been active in a long time, so I do not know the current state of things. I have no answers for your question.

Instead, ask using one of the [communication channels](https://www.flashrom.org/contact.html), or use the [ticket system](https://ticket.coreboot.org/projects/flashrom) if you want to report an issue.

closed 2024-08-07 by Th3Fanbus