flashrom/flashrom #144: support for XT25F128

orangecms, 2020-05-01:
from flashrom:

```
flashrom v1.2 on Linux 4.19.72-gentoo-9e (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.6, GCC 9.3.0, little endian
Command line (3 args): flashrom -p ch341a_spi -VV
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing ch341a_spi programmer
Device revision is 3.0.4
The following protocols are supported: SPI.
Probing for AMIC A25L010, 128 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for AMIC A25L016, 2048 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
...
Probing for Unknown SFDP-capable chip, 0 kB: SFDP revision = 1.0
SFDP number of parameter headers is 2 (NPH = 1).

SFDP parameter table header 0/1:
  ID 0x00, version 1.0
  Length 36 B, Parameter Table Pointer 0x000030
Parsing JEDEC flash parameter table...
  3-Byte only addressing.
  Status register is non-volatile and the standard does not allow vendors to tell us whether EWSR/WREN is needed for status register writes - assuming EWSR.
  Write chunk size is at least 64 B.
  Flash chip size is 16384 kB.
  Block eraser 0: 4096 x 4096 B with opcode 0x20
  Tried to add a duplicate block eraser: 4096 x 4096 B with opcode 0x20.
  Block eraser 1: 512 x 32768 B with opcode 0x52
  Block eraser 2: 256 x 65536 B with opcode 0xd8
done.

SFDP parameter table header 1/1:
  ID 0x0b, version 1.0
  Length 12 B, Parameter Table Pointer 0x000060
===
SFDP has autodetected a flash chip which is not natively supported by flashrom yet.
All standard operations (read, verify, erase and write) should work, but to support all possible features we need to add them manually.
You can help us by mailing us the output of the following command to flashrom@flashrom.org:
'flashrom -VV [plus the -p/--programmer parameter]'
Thanks for your help!
===
Found Unknown flash chip "SFDP-capable chip" (16384 kB, SPI) on ch341a_spi.
Probing for AMIC unknown AMIC SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Atmel unknown Atmel SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Eon unknown Eon SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Macronix unknown Macronix SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for PMC unknown PMC SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for SST unknown SST SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for ST unknown ST SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Sanyo unknown Sanyo SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Winbond unknown Winbond (ex Nexcom) SPI chip, 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Generic unknown SPI chip (RDID), 0 kB: probe_spi_rdid_generic: id1 0x0b, id2 0x4018
Probing for Generic unknown SPI chip (REMS), 0 kB: probe_spi_rems: id1 0xb, id2 0x17
Found Unknown flash chip "SFDP-capable chip" (16384 kB, SPI).
No operations were specified.
```

see also https://patchwork.ozlabs.org/project/linux-mtd/patch/20200130085512.GA243814@makrotopia.org/:

```
/* XTX Technology (Shenzhen) Limited */
"xt25f128a", INFO(0x207018, 0, 64 * 1024, 256, SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ) },
"xt25f128b", INFO(0x0B4018, 0, 64 * 1024, 256, SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ) },
```

orangecms, 2020-05-01:
Reading and writing both went fine when I tried. This SPI flash can be found on the LicheePi Nano. :)

sleeply4cat, 2020-08-06:
For other visitors: this chip doesn't work with 3.3v, you need an adapter for 1.8v.