flashrom/flashrom #145: ST M95640

greg2891, 2020-05-01:
Does flashrom support this chip ST M95640 - trying to test but always Not Found
Thanks

dhendrix, 2020-05-02:
There is an open PR for ST 95xxx support, however it's not upstream yet: https://github.com/flashrom/flashrom/pull/84

There was also a report of success: https://mail.coreboot.org/hyperkitty/list/flashrom@flashrom.org/message/SSJSHQ4CBHFGBXLI4LA7QDGEDREOAZ3B/

Hopefully somebody will have time to complete the patch and merge it.

evrinoma, 2020-11-02:
Can somebody help me with this issue?

greg2891, 2021-03-20:
i have no idea why the patch for M95640 was not included yet - @evrinoma  helped me previously and saved my life :) with his patch - please can anyone let us know why M95640 i still not in flashrom

Idontker, 2025-09-29:
Hi there, I was wondering, if and how I can help out with this feature. If I read it correctly, there is a patch for the M95xxxx series. What work is missing? 
Best regards :)