flashrom/flashrom #148: Flashrom support for XM25QH64AHIG

ayushman4, 2020-07-01:
I am trying to read the flash chip using a ch341a programmer and I get the following response:

Found ST Flash chip "Unknown ST SPI chip" (0 kb, SPI) on ch341a_spi.

This Flash has status NOT WORKING for operations: PROBE READ ERASE WRITE.

The datasheet is https://datasheet.lcsc.com/szlcsc/1811072025_XMC-XM25QH64AHIG_C328461.pdf

jaggu1989, 2024-08-19:
kindly add support for W25Q128FWSQ
datasheet
https://www.alldatasheet.com/view.jsp?Searchword=25q128fwsq%20datasheet&gad_source=1&gclid=EAIaIQobChMI5brtpNSBiAMVve0WBR3nrydHEAAYAiAAEgJzyfD_BwE

ericswpark, 2024-10-23:
@felixsinger why were the two PRs closed? I tried to access the Gerrit link but it's dead, or I don't have sufficient permissions to view it. I see on the latest definition file support for XM25QH64C, but not A.

ericswpark, 2025-03-30:
Support for this chip was added yesterday with the following patchset on Gerrit: https://review.coreboot.org/c/flashrom/+/86990

@ayushman4 would you please close off this issue? The flashrom team doesn't use GitHub, and as I'm not the original reporter I can't close it on your behalf. Thanks!