flashrom/flashrom #149: Can´t compile with make CC=i686-w64-mingw32-gcc

BaltasarParreira, 2020-07-02:
Hi, having some strange compile error compiling in windows 10 using cygwin/mingw.
I got the following error:

`i686-w64-mingw32-gcc -MMD -Os -Wall -Wextra -Wno-unused-parameter -Wshadow -Wmissing-prototypes -Wwrite-strings -Werror -I/usr/include/libusb-1.0  -Dffs=__builtin_ffs -D__USE_MINGW_ANSI_STDIO=1 -D'CONFIG_DEFAULT_PROGRAMMER=PROGRAMMER_INVALID' -D'CONFIG_DEFAULT_PROGRAMMER_ARGS="''"' -D'CONFIG_SERPROG=1' -D'CONFIG_PONY_SPI=1' -D'CONFIG_BITBANG_SPI=1'   -D'CONFIG_PICKIT2_SPI=1' -D'CONFIG_STLINKV3_SPI=1'  -I/usr/include/libftdi1 -I/usr/include/libusb-1.0  -D'CONFIG_DUMMY=1' -D'CONFIG_BUSPIRATE_SPI=1' -D'CONFIG_DEDIPROG=1' -D'CONFIG_DEVELOPERBOX_SPI=1' -D'CONFIG_CH341A_SPI=1' -D'CONFIG_DIGILENT_SPI=1' -D'NEED_LIBUSB1=1'   -D'FLASHROM_VERSION="v1.2"' -o serial.o -c serial.c
serial.c: In function 'serialport_write':
serial.c:398:11: error: comparison of integer expressions of different signedness: 'DWORD' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
  398 |   if (tmp == -1) {
      |           ^~
serial.c: In function 'serialport_read':
serial.c:432:11: error: comparison of integer expressions of different signedness: 'DWORD' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
  432 |   if (tmp == -1) {
      |           ^~
serial.c: In function 'serialport_read_nonblock':
serial.c:494:11: error: comparison of integer expressions of different signedness: 'DWORD' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
  494 |   if ((rv == -1) && (errno != EAGAIN)) {
      |           ^~
serial.c: In function 'serialport_write_nonblock':
serial.c:574:11: error: comparison of integer expressions of different signedness: 'DWORD' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
  574 |   if ((rv == -1) && (errno != EAGAIN)) {
      |           ^~
cc1: all warnings being treated as errors
make: *** [Makefile:1146: serial.o] Error 1`

dhendrix, 2020-07-02:
Thanks for the report. The error isn't strange :-) There simply seems to be some invalid comparisons done on return values - Windows returns booleans while POSIX returns signed ints.

I whipped up a patch, but have not tested it since I don't have a Windows system handy. Give this a try and LMK if it works for you: https://github.com/dhendrix/flashrom/commit/b8758052f2f3adbd3e456a5d32e7421505ab8078

BaltasarParreira, 2020-07-02:
Thanks, I downloaded the patched version and It worked for the serial.c error, but further ahead crashed again with this bigger one:
`i686-w64-mingw32-gcc -Os -Wall -Wshadow -Werror  -MMD -MP -MF .dep/ich_descriptors.o.d -D ICH_DESCRIPTORS_FROM_DUMP_ONLY -I../../ -I. -o .obj/ich_descriptors.o -c ../../ich_descriptors.c
../../ich_descriptors.c: In function 'prettyprint_ich_descriptor_content':
../../ich_descriptors.c:139:12: error: unknown conversion type character 'z' in format [-Werror=format=]
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:139:2: note: in expansion of macro 'msg_pdbg2'
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:139:64: note: format string is defined here
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |                                                                ^
../../ich_descriptors.c:139:12: error: too many arguments for format [-Werror=format-extra-args]
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:139:2: note: in expansion of macro 'msg_pdbg2'
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:139:12: error: unknown conversion type character 'z' in format [-Werror=format=]
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:139:2: note: in expansion of macro 'msg_pdbg2'
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:139:64: note: format string is defined here
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |                                                                ^
../../ich_descriptors.c:139:12: error: too many arguments for format [-Werror=format-extra-args]
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:139:2: note: in expansion of macro 'msg_pdbg2'
  139 |  msg_pdbg2("NR          (Number of Regions):                 %5zd\n",   ich_number_of_regions(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:145:12: error: unknown conversion type character 'z' in format [-Werror=format=]
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:145:2: note: in expansion of macro 'msg_pdbg2'
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:145:64: note: format string is defined here
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |                                                                ^
../../ich_descriptors.c:145:12: error: too many arguments for format [-Werror=format-extra-args]
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../ich_descriptors.c:23:30: note: in definition of macro 'print'
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:145:2: note: in expansion of macro 'msg_pdbg2'
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |  ^~~~~~~~~
../../ich_descriptors.c:145:12: error: unknown conversion type character 'z' in format [-Werror=format=]
  145 |  msg_pdbg2("NM          (Number of Masters):                 %5zd\n",   ich_number_of_masters(cs, cont));
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
   23 | #define print(t, ...) printf(__VA_ARGS__)
      |                              ^~~~~~~~~~~
../../ich_descriptors.c:397:3: note: in expansion of macro 'msg_pdbg2'
  397 |   msg_pdbg2("FLMSTR%zd  0x%08x\n", i + 1, desc->master.FLMSTRs[i]);
      |   ^~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:66: .obj/ich_descriptors.o] Error 1
make[1]: Leaving directory '/home/Balta/flashrom-b8758052f2f3adbd3e456a5d32e7421505ab8078/util/ich_descriptors_tool'
make: *** [Makefile:1201: all] Error 2`

dhendrix, 2020-07-02:
Darn... It looks like there are going to be a lot more errors to work thru in order to get compilation working with this particular setup :-/

I can't look at it now, unfortunately, but we can leave this open and hope somebody else can pick it up.

BaltasarParreira, 2020-07-02:
I see.
Is there any other way to compile or cross compile a windows version?

dhendrix, 2020-07-02:
I looked into this a bit more and the fix is actually quite easy - We just need to set __USE_MINGW_ANSI_STDIO for ich_descriptors_tool. The proper fix is in the Makefile here: https://github.com/dhendrix/flashrom/commit/4f029d9127da25e44b287b4f0eab8b7d6f0ede83 .  With that I can run `make CC=i686-w64-mingw32-gcc CONFIG_ENABLE_LIBUSB1_PROGRAMMERS=no` on my Ubuntu system.

However, as a stop-gap you can add CFLAGS+=-D__USE_MINGW_ANSI_STDIO=1 to your compilation command.

(BTW - check out https://flashrom.org/Windows for more info on compiling flashrom for Windows)

BaltasarParreira, 2020-07-03:
Yeeaahhhhh !!!
It worked like a charm.
Thanks for the amazing support.

dhendrix, 2020-07-03:
Great, glad to know it worked! I'll send the patches upstream and close this out once they land.

dhendrix, 2020-07-03:
Patches have been sent upstream for review:
https://review.coreboot.org/c/flashrom/+/43051
https://review.coreboot.org/c/flashrom/+/43052

BaltasarParreira, 2020-07-03:
Hi, I just realized now that my compiled version doesn't support FTDI.
Recompiled it again and I noted that gives this in the beginning:
`$ make CC=i686-w64-mingw32-gcc CFLAGS+=-D__USE_MINGW_ANSI_STDIO=1
make: ./util/getrevision.sh: No such file or directory
make: ./util/getrevision.sh: No such file or directory
Replacing all version templates with .
Checking for a C compiler... found.
Target arch is x86
Target OS is MinGW
Checking for libusb-1.0 headers... found.
Checking if libusb-1.0 is usable... yes.
**Checking for FTDI support... not found.**
Checking for utsname support... not found.
Checking for clock_gettime support... not found.`

Is there a way to solve this?
I checked and I have all the required FTDI lib files.

dhendrix, 2020-07-03:
Give this a try: https://flashrom.org/Windows#libftdi

BaltasarParreira, 2020-07-03:
Thanks, but that was what I followed since the beginning.
That's why I said that I have all the libs on the right place.
I think I'm going to do a fresh install since right now with all the tests I did and changed so many things that don't really know what files, versions and where they are.

BaltasarParreira, 2020-07-04:
Ok I leave here the solution has anyone could need it.

The solution is having on mingw the PKG-CONFIG tool working and for that we have to do this:

```
First of all, you must download [pkg-config.exe](http://ftp.gnome.org/pub/gnome/binaries/win32/dependencies/pkg-config_0.23-3_win32.zip) from GTK+ for Windows.
Then you can simply unzip this downloaded zip-file, and then copy the pkg-config.exe executable into /MinGW/bin

That's not enough: this executable depends on the GLib DLL.
So you must download GLib DLL too, always from GTK+ for Windows.
Once again, you have to unzip this downloaded zip-file, and then copy liglib-2.0-0.dll into /MinGW/bin

You have not yet finished: pkg-config still has an unresolved DLL dependency.
But this time simply performing a trivial copy will suffit. So you must now open an MSYS shell:

cd C:/MinGW/bin
cp libintl-8.dll intl.dll

And this time that's really all: now you have pkg-confif properly installed and ready to work.
```

dhendrix, 2020-07-04:
Thanks for writing that up! I've added some info about pkg-config and a link to your comment in the flashrom wiki: https://flashrom.org/Windows#MinGW.2FMSYS

BTW, it seems there might be an easier way to install pkg-config using `pacman` with MSYS2: https://stackoverflow.com/a/26476504/8012621

Is that a good way of doing it? Maybe we should update the flashrom wiki to suggest using MSYS2 if possible since it seems a lot simpler.

BaltasarParreira, 2020-07-04:
Yes MSYS2 also has they one packages for the LIBUSB and LIBFTDI easy to direct install.
I did like that since I´m using mingw32 as all the installers out there for the Mingw64 with MSYS2 report some sort of virus or something malware and I couldn't find any valid information that even from the original site that this is some sort of false positive. ;)

obskyr, 2022-02-16:
I'm trying to build version 1.2 on Windows using MSYS2 – I followed the instructions on the wiki to the dot, running the command `make CONFIG_ENABLE_LIBPCI_PROGRAMMERS=no` (as simply `make` gave me the error `Checking for libpci headers... not found.` – is that right?). However, I consistently get the following error:

```
make: ./util/getrevision.sh: No such file or directory
make: ./util/getrevision.sh: No such file or directory
Replacing all version templates with .
Checking for a C compiler... found.
Target arch is x86
Target OS is Cygwin
Checking for libusb-1.0 headers... not found.
```

Looking into it, the compiler can't find `libusb.h` since it's named `lusb0_usb.h`, but renaming it to `libusb.h` instead gives the following error in the test command `cc -c -I/usr/include/libusb-1.0 -I/usr/local/include -Werror .test.c -o .test.o`:

```
c -o .atest.o
.test.c: In function ‘main’:
.test.c:7:9: error: implicit declaration of function ‘libusb_init’; did you mean ‘usb_init’? [-Werror=implicit-function-declaration]
    7 |         libusb_init(NULL);
      |         ^~~~~~~~~~~
      |         usb_init
cc1: all warnings being treated as errors
```

It's apparent that I have some sort of library installation issues, but I don't know exactly what they are! Might someone know how I can fix this and successfully build on Windows?