flashrom/flashrom #153: [REQUEST] Adding Puya's P25Q40H to the list of supported devices.

KaoDome, 2020-07-18:
Hi, from what I could see, there is no support for the P25Q40H found in some external SATA III to USB 3.0 enclosures. I would like to request its support in Flashrom if at all possible.

I'm attaching the data sheet for that particular model and some others from the same series.

Thank you in any case, kind regards.

[P25Q40H-Puya.pdf](https://github.com/flashrom/flashrom/files/4941238/P25Q40H-Puya.pdf)

KaoDome, 2020-07-21:
I'm attaching the output of the -VV execution for both chips since reading from them was possible:

[VV-PUYA-P25D21H.txt](https://github.com/flashrom/flashrom/files/4955198/VV-PUYA-P25D21H.txt)
[VV-PUYA-P25Q40H.txt](https://github.com/flashrom/flashrom/files/4955199/VV-PUYA-P25Q40H.txt)

Th3Fanbus, 2020-07-26:
You can also try adding the flash chips yourself! It's not that difficult. I'm bad at explaining, but I hope this will be enough.

For the P25Q40H log, I see this:

```
SFDP parameter table header 0/1:
  ID 0x00, version 1.0
  Length 36 B, Parameter Table Pointer 0x000030
Parsing JEDEC flash parameter table... 
  3-Byte only addressing.
  Status register is non-volatile and the standard does not allow vendors to tell us whether EWSR/WREN is needed for status register writes - assuming EWSR.
  Write chunk size is at least 64 B.
  Flash chip size is 512 kB.
  Block eraser 0: 128 x 4096 B with opcode 0x20
  Tried to add a duplicate block eraser: 128 x 4096 B with opcode 0x20.
  Block eraser 1: 16 x 32768 B with opcode 0x52
  Block eraser 2: 8 x 65536 B with opcode 0xd8
  Block eraser 3: 2048 x 256 B with opcode 0x81
done.
```

This is SFDP (Serial Flash Discovery Parameter) info. That is, the flash chip has a built-in quick starter guide, which flashrom used to read the flash chip.

It tells you that this chip does not support 4BA (4-byte addressing). Then, the block erasers. When adding them to flashrom, sort them in erase size ascending order:

```
  Block eraser 3: 2048 x 256 B with opcode 0x81
  Block eraser 0: 128 x 4096 B with opcode 0x20
  Block eraser 1: 16 x 32768 B with opcode 0x52
  Block eraser 2: 8 x 65536 B with opcode 0xd8
```

This also tells us that the "page_size" parameter should be 256 (like on nearly every other SPI flash chip). Write chunk size is at least 64B, so 256 is fine. We need to erase before writing, anyway. For example, the first two erasers would look like this:

```
{
	.eraseblocks = { {256, 2048} },
	.block_erase = spi_block_erase_81,
}, {
	.eraseblocks = { {4 * 1024, 128} },
	.block_erase = spi_block_erase_20,
}
```

You should be able to easily figure out the other two eraser definitions.

```
Probing for Generic unknown SPI chip (RDID), 0 kB: probe_spi_rdid_generic: id1 0x85, id2 0x6013
```

This line contains the vendor ID (0x85) and the device ID (0x6013). Yes, it appears nearly everywhere, because we don't cache it. [1]

You can compare this information with the datasheet and then add a new entry to `flashchips.c`. Then, push a patch to Gerrit. The GitHub repo is a mirror from Gerrit, where we review and submit patches. We use the same infrastructure as coreboot, so you can use the same instructions: https://doc.coreboot.org/tutorial/part2.html

In case you get an error when running `git push`, try with `git push origin HEAD:refs/for/master` instead.

For reference, here's an example commit that adds a flash chip: https://review.coreboot.org/43573

[1]: I wanted to change that, but I realized that it's a good way to determine if the connection with the flash chip is reliable. In your case, the IDs are always the same, which is good. Compare that to https://paste.flashrom.org/view.php?id=3338 where the IDs are sometimes all-zeroes or all-ones (0xff or 0xffff).