flashrom/flashrom #157: Run Command Failed in Linux

wangzitiansky, 2020-07-19:
I run the command **sudo flashrom -p internal** Failed in Ubuntu. 
```bash
~/Desktop » sudo flashrom -p internal                                        sky@sky-virtual-machine
flashrom v1.2 on Linux 5.4.0-40-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel PIIX4/4E/4M".
WARNING: No chipset found. Flash detection will most likely fail.
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.

```
Want to know why? Can anybody help me? Thanks!!

Th3Fanbus, 2020-07-26:
Are you running flashrom in a virtual machine?

wangzitiansky, 2020-07-26:
Yes! flashrom can not run in a virtual machine?

Th3Fanbus, 2020-07-26:
No, the internal programmer won't work in a virtual machine. However, USB programmers should work if you can connect the USB programmer to the virtual machine.