flashrom/flashrom #164: support for XT25F64B

orangecms, 2020-08-19:
This chip is found on an IPC-RB-BLK530AI board (IP camera), which I took apart to read its name and tried to dump.

I can't read from the chip too reliably while in circuit. I mostly get the message that no chip is found, but sometimes
> Found Generic flash chip "unknown SPI chip (RDID)" (0 kB, SPI) on ch341a_spi.

For the datasheet, see http://file2.dzsc.com/product/19/06/22/216185_132959081.pdf

#98 look like it's a chip from the same vendor.

I am using a CH341A programmer.

Here is a log from what seems to return "something", don't know if it's useful in any way.

```
flashrom v1.2 on Linux 5.7.12-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.6.4, GCC 9.3.0, little endian
Command line (3 args): flashrom -p ch341a_spi -VVV
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing ch341a_spi programmer
Device revision is 3.0.4
Wrote 3 bytes:
 aa 61 00

Wrote 4 bytes:
 ab b7 7f 20

The following protocols are supported: SPI.
Probing for AMIC A25L010, 128 kB: programmer_map_flash_region: mapping A25L010 from 0x00000000fffe0000 to 0x0000000000000000
Wrote 37 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 f9 ff ff ff

Read 4 bytes:
 9c eb f0 f5

RDID returned 0xd7 0x0f 0xaf. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0xd7, id2 0xfaf
programmer_unmap_flash_region: unmapped 0x0000000000000000
Probing for AMIC A25L016, 2048 kB: programmer_map_flash_region: mapping A25L016 from 0x00000000ffe00000 to 0x0000000000000000
Wrote 37 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 f9 ff ff ff

Read 4 bytes:
 ff ff ff ff

RDID returned 0xff 0xff 0xff. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0xff, id2 0xffff
programmer_unmap_flash_region: unmapped 0x0000000000000000
Probing for AMIC A25L020, 256 kB: programmer_map_flash_region: mapping A25L020 from 0x00000000fffc0000 to 0x0000000000000000
Wrote 37 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 f9 ff ff ff

Read 4 bytes:
 00 00 00 00

RDID returned 0x00 0x00 0x00. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0x00, id2 0x00
programmer_unmap_flash_region: unmapped 0x0000000000000000
Probing for AMIC A25L032, 4096 kB: programmer_map_flash_region: mapping A25L032 from 0x00000000ffc00000 to 0x0000000000000000
Wrote 37 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 f9 ff ff ff

Read 4 bytes:
 00 00 00 00

RDID returned 0x00 0x00 0x00. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0x00, id2 0x00
programmer_unmap_flash_region: unmapped 0x0000000000000000
Probing for AMIC A25L040, 512 kB: programmer_map_flash_region: mapping A25L040 from 0x00000000fff80000 to 0x0000000000000000
[... 3584 lines cut ...]
 ff ff ff ff

RDID returned 0xff 0xff 0xff. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0xff, id2 0xffff
Probing for Generic unknown SPI chip (RDID), 0 kB: Wrote 37 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 f9 ff ff ff

Read 4 bytes:
 ff ff ff ff

RDID returned 0xff 0xff 0xff. RDID byte 0 parity violation. probe_spi_rdid_generic: id1 0xff, id2 0xffff
Probing for Generic unknown SPI chip (REMS), 0 kB: Wrote 39 bytes:
 ab b7 b7 b7 b6 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 a8 09 00 00 00 ff ff

Read 6 bytes:
 ff ff ff ff ff ff

REMS returned 0xff 0xff. probe_spi_rems: id1 0xff, id2 0xff
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
Wrote 4 bytes:
 ab b7 40 20

```

LEENO82, 2020-10-15:
Same problem for me
eeprom in ipcam,
interrupted while updating.
Need to rewrite bin

Marantral, 2020-10-22:
I am not able to dump that type of flash. Looking at the supported hardware page, it looks to not be supported. https://flashrom.org/Supported_hardware.