flashrom/flashrom #168: Build failed : mecl308_init

d51r3verse, 2020-09-19:
```
root@Orange:/opt/spi/flashrom# git log -1
commit 702c58a3218ee2c953b053d06c29ee623d16cd01
Author: Nikolai Artemiev <nartemiev@google.com>
Date:   Wed Sep 16 14:12:22 2020 +1000

    tests: Update licence headers

    Signed-off-by: Nikolai Artemiev <nartemiev@google.com>
    Change-Id: Ia964279ace569b4b93f4e2919c1c228a9b621745
    Reviewed-on: https://review.coreboot.org/c/flashrom/+/45438
    Tested-by: build bot (Jenkins) <no-reply@coreboot.org>
    Reviewed-by: Patrick Georgi <pgeorgi@google.com>
    Reviewed-by: Edward O'Callaghan <quasisec@chromium.org>
    Reviewed-by: Angel Pons <th3fanbus@gmail.com>
root@Orange:/opt/spi/flashrom# make CONFIG_ENABLE_LIBPCI_PROGRAMMERS=no  CONFIG_ENABLE_LIBUSB1_PROGRAMMERS=no
Replacing all version templates with v1.2-105-g702c58a.
Checking for a C compiler... found.
Target arch is arm
Target OS is Linux
Checking for FTDI support... found.
Checking for FT232H support in libftdi... found.
Checking if Linux MTD headers are present... yes.
Checking if Linux SPI headers are present... yes.
Checking for utsname support... found.
Checking for clock_gettime support... found.
...
...(snip cc)
...
flashrom.c:92:13: error: 'mec1308_init' undeclared here (not in a function)
   .init   = mec1308_init,
             ^~~~~~~~~~~~
Makefile:1267: recipe for target 'flashrom.o' failed
make: *** [flashrom.o] Error 1
```

jamesnorth, 2020-09-28:
I get the same issue with the following commit.
commit d3b6acffe40351d84cfd6aaa0238f9a653d69aa1

septs, 2020-10-26:
> The fixes are now at: https://review.coreboot.org/c/flashrom/+/46811 and https://review.coreboot.org/c/flashrom/+/46812.

from https://github.com/flashrom/flashrom/issues/172#issuecomment-716372980

plz close this issue.

closed 2020-10-27 by d51r3verse