flashrom/flashrom #172: mec1308.c build failed

septs, 2020-10-24:
```
septs@CBDesktop02 ~ $ brew install flashrom --HEAD
Updating Homebrew...
==> Cloning https://review.coreboot.org/flashrom.git
Updating /Users/septs/Library/Caches/Homebrew/flashrom--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 06ca247 flashrom.c,flash.h: Kill dead fn shutdown_free()
==> make DESTDIR=/usr/local/Cellar/flashrom/HEAD-06ca247 PREFIX=/ install
Last 15 lines from /Users/septs/Library/Logs/Homebrew/flashrom/01.make:
2 errors generated.
make: *** [ene_lpc.o] Error 1
make: *** Waiting for unfinished jobs....
mec1308.c:512:2: error: use of undeclared identifier 'internal_buses_supported'; did you mean 'get_buses_supported'?
        internal_buses_supported |= BUS_LPC;    /* for LPC <--> SPI bridging */
        ^~~~~~~~~~~~~~~~~~~~~~~~
        get_buses_supported
./flash.h:424:18: note: 'get_buses_supported' declared here
enum chipbustype get_buses_supported(void);
                 ^
mec1308.c:512:27: error: invalid operands to binary expression ('enum chipbustype (void)' and 'int')
        internal_buses_supported |= BUS_LPC;    /* for LPC <--> SPI bridging */
        ~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~
2 errors generated.
make: *** [mec1308.o] Error 1

READ THIS: https://docs.brew.sh/Troubleshooting

Please create pull requests instead of asking for help on Homebrew's GitHub,
Discourse, Twitter or IRC.
```

follow commit: https://github.com/flashrom/flashrom/commit/821e44cb4dc23ca1261a92b3ee3cdf591c1da91b ?

septs, 2020-10-24:
CC @0dvictor request fix the it.

Th3Fanbus, 2020-10-24:
Builds fine for me on Arch Linux with GCC 10. Which compiler are you using?

septs, 2020-10-24:
```
septs@CBDesktop02 ~ $ gcc -v
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.32.21)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

0dvictor, 2020-10-26:
I'm unable to reproduce it on my Debian based Linux with clang 12. I'll find a mac and try it out.

septs, 2020-10-26:
@0dvictor homebrew pre-disable some feature.

```ruby
    ENV["CONFIG_GFXNVIDIA"] = "0"
    ENV["CONFIG_NIC3COM"] = "0"
    ENV["CONFIG_NICREALTEK"] = "0"
    ENV["CONFIG_NICNATSEMI"] = "0"
    ENV["CONFIG_NICINTEL"] = "0"
    ENV["CONFIG_NICINTEL_SPI"] = "0"
    ENV["CONFIG_NICINTEL_EEPROM"] = "0"
    ENV["CONFIG_OGP_SPI"] = "0"
    ENV["CONFIG_SATAMV"] = "0"
    ENV["CONFIG_SATASII"] = "0"
    ENV["CONFIG_DRKAISER"] = "0"
    ENV["CONFIG_RAYER_SPI"] = "0"
    ENV["CONFIG_INTERNAL"] = "0"
    ENV["CONFIG_IT8212"] = "0"
    ENV["CONFIG_ATAHPT"] = "0"
    ENV["CONFIG_ATAVIA"] = "0"
```

see https://github.com/Homebrew/homebrew-core/blob/c3f9b20738958c3f88ef165b5020c4743302311a/Formula/flashrom.rb#L27-L42

septs, 2020-10-26:
```
septs@binwalk:~/flashrom$ CC=clang CONFIG_GFXNVIDIA=0 CONFIG_NIC3COM=0 CONFIG_NICREALTEK=0 CONFIG_NICNATSEMI=0 CONFIG_NICINTEL=0 CONFIG_NICINTEL_SPI=0 CONFIG_NICINTEL_EEPROM=0 CONFIG_OGP_SPI=0 CONFIG_SATAMV=0 CONFIG_SATASII=0 CONFIG_DRKAISER=0 CONFIG_RAYER_SPI=0 CONFIG_INTERNAL=0 CONFIG_IT8212=0 CONFIG_ATAHPT=0 CONFIG_ATAVIA=0 make
Replacing all version templates with v1.2-121-g06ca247.
Checking for a C compiler... found.
Target arch is x86
Target OS is Linux
Checking for libusb-1.0 headers... found.
Checking if libusb-1.0 is usable... yes.
Checking for FTDI support... not found.
Checking if Linux MTD headers are present... yes.
Checking if Linux SPI headers are present... yes.
Checking for utsname support... found.
Checking for clock_gettime support... found.
clang -MMD -Os -Wall -Wextra -Wno-unused-parameter -Wshadow -Wmissing-prototypes -Wwrite-strings -Werror -I/usr/include/libusb-1.0  -D'CONFIG_DEFAULT_PROGRAMMER=PROGRAMMER_INVALID' -D'CONFIG_DEFAULT_PROGRAMMER_ARGS="''"' -D'CONFIG_ENE_LPC=1' -D'CONFIG_MEC1308=1' -D'CONFIG_SERPROG=1' -D'CONFIG_RAIDEN=1' -D'CONFIG_PONY_SPI=1' -D'CONFIG_BITBANG_SPI=1'   -D'CONFIG_PICKIT2_SPI=1' -D'CONFIG_STLINKV3_SPI=1'   -D'CONFIG_DUMMY=1' -D'CONFIG_BUSPIRATE_SPI=1' -D'CONFIG_DEDIPROG=1' -D'CONFIG_DEVELOPERBOX_SPI=1' -D'CONFIG_LINUX_MTD=1' -D'CONFIG_LINUX_SPI=1' -D'CONFIG_CH341A_SPI=1' -D'CONFIG_DIGILENT_SPI=1' -D'CONFIG_I2C_SUPPORT=1' -D'NEED_LIBUSB1=1' -D'HAVE_UTSNAME=1' -D'HAVE_CLOCK_GETTIME=1' -D'FLASHROM_VERSION="v1.2-121-g06ca247"' -o ene_lpc.o -c ene_lpc.c
ene_lpc.c:167:9: error: implicit declaration of function 'INB' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        while (INB(chip->port_ec_command) & MASK_INPUT_BUFFER_FULL) {
               ^
ene_lpc.c:176:2: error: implicit declaration of function 'OUTB' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        OUTB(cmd, chip->port_ec_command);
        ^
ene_lpc.c:584:2: error: use of undeclared identifier 'internal_buses_supported'; did you mean 'get_buses_supported'?
        internal_buses_supported |= BUS_LPC;
        ^~~~~~~~~~~~~~~~~~~~~~~~
        get_buses_supported
./flash.h:424:18: note: 'get_buses_supported' declared here
enum chipbustype get_buses_supported(void);
                 ^
ene_lpc.c:584:27: error: invalid operands to binary expression ('enum chipbustype (void)' and 'int')
        internal_buses_supported |= BUS_LPC;
        ~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~
4 errors generated.
Makefile:1267: recipe for target 'ene_lpc.o' failed
make: *** [ene_lpc.o] Error 1
```

----------

Platform: Ubuntu 16.04.7 LTS

septs, 2020-10-26:
make sense, i think maybe need a `ene_lpc` and `mec1308` disable flag for macOS user.

septs, 2020-10-26:
https://github.com/flashrom/flashrom/blob/b1f858f65b2abd276542650d8cb9e382da258967/programmer.h#L356-L365

@0dvictor caused by `CONFIG_INTERNAL=0`, so build failed.

0dvictor, 2020-10-26:
I am able to reproduce [this](https://github.com/flashrom/flashrom/issues/172#issuecomment-716317249) on my Linux box.

It showed two problems:
1. `NEED_RAW_ACCESS` is missing. I created #173 to fix it.
2. Access `internal_buses_supported` probably should be guarded with `#ifdef CONFIG_INTERNAL`, which I am still working on.

septs, 2020-10-26:
@0dvictor i think, add `CONFIG_ENE_LPC` and `CONFIG_MEC1308` to `CONFIG_ENABLE_LIBPCI_PROGRAMMERS` control disable it.

0dvictor, 2020-10-26:
Good suggestion, let me do that.

septs, 2020-10-26:
i think, `Build Bot` need support verify for `CONFIG_ENABLE_LIBUSB0_PROGRAMMERS=no`, `CONFIG_ENABLE_LIBUSB0_PROGRAMMERS=no` and `CONFIG_ENABLE_LIBPCI_PROGRAMMERS=no` build mode.

but i don’t know who the problem with ?

0dvictor, 2020-10-26:
I've updated #173 to include both fixes. I'll send the same patches to https://review.coreboot.org/q/project:flashrom.

0dvictor, 2020-10-26:
The fixes are now at: https://review.coreboot.org/c/flashrom/+/46811 and https://review.coreboot.org/c/flashrom/+/46812.

closed 2020-10-26 by septs

nobk, 2020-11-05:
@0dvictor  Compile on Raspbian on Raspberry Pi 2 still got this error, after your fixes.
```
make -j1
Replacing all version templates with v1.2-135-g07edbb9.
Checking for a C compiler... found.
Target arch is arm
Target OS is Linux
Checking for libpci headers... found.
Checking version of pci_get_dev... new version (including PCI domain parameter).
Checking if libpci is present and sufficient... yes.
Checking for libusb-1.0 headers... found.
Checking if libusb-1.0 is usable... yes.
Checking for FTDI support... found.
Checking for FT232H support in libftdi... found.
Checking if Linux MTD headers are present... yes.
Checking if Linux SPI headers are present... yes.
Checking for utsname support... found.
Checking for clock_gettime support... found.
......
flashrom.c:92:13: error: 'mec1308_init' undeclared here (not in a function); did you mean 'pci_init'?
   .init   = mec1308_init,
             ^~~~~~~~~~~~
             pci_init

```

0dvictor, 2020-11-11:
Apologize for the inconvenience, this patch should fix it. (Issue #175)
https://review.coreboot.org/c/flashrom/+/47394