flashrom/flashrom #176: meson: c_std=gnu99 is required on PPC64

mkszuba, 2020-11-11:
The current version of meson.build sets c_std=c99. This results in compilation errors (with, at least, gcc-9 and gcc-10) in hwaccess.c on PPC64 (as well as possibly other arches, not on amd64 though) due to the compiler not recognising the function _asm()_. In order for PPC64 compilation to succeed it is necessary to change c_std=gnu99 or higher.

mkszuba, 2020-11-11:
Update: I've just noticed #155 , which might be a better solution (chances are c_std=gnu99 will not help while linking against musl or other non-GNU standard library).

Th3Fanbus, 2020-11-11:
For compatibility reasons, flashrom should only use ISO C. If flashrom code uses GNU extensions, then it's the code that needs to be fixed.

neheb, 2022-03-13:
asm() is not ISO C