flashrom/flashrom #177: Not working - Big Sur or Big Sur beta

JoeOIVOV, 2020-11-27:
Not sure if this a Flashrom problem, or a problem with Xcode 12.2 or 12.2 beta,  but Flashrom doesn't detect the SPI reader on Big Sur. 
____________________________________________________________
flashrom v1.2 on Darwin 20.1.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Couldn't open device 1a86:5512.
Error: Programmer initialization failed.
____________________________________________________________

I switch to High Sierra and it works as expected without changing anything other than OS.

JoeOIVOV, 2020-12-11:
Seems to been fixed by OpenCore 0.6.4.  

sudo flashrom -p ch341a_spi
Password:
flashrom v1.2 on Darwin 20.1.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
libusb: error [darwin_claim_interface] interface not found
Failed to claim interface 0: 'LIBUSB_ERROR_NOT_FOUND'
Error: Programmer initialization failed.

sudo flashrom -p ch341a_spi
flashrom v1.2 on Darwin 20.1.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
administrator@Administrators-Mac-Pro ~ % sudo flashrom -p ch341a_spi
flashrom v1.2 on Darwin 20.1.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

closed 2020-12-11 by JoeOIVOV

McMac, 2021-02-14:
How is this closed? This only works once. If you disconnect the reader from the rom you are trying to flash (e.g. TitanRidge has two of them) it fails again. 
In other words, connect everything. If you somehow have a bad connection and need to reseat the connector, well, guess what, reboot once again.