flashrom/flashrom #182: Unhandled programmer parameters: dualbiosindex?

zailawee, 2021-01-07:
I'm trying to use flashrom 1.2 on a GIGABYTE GA-H270N-WIFI motherboard with DualBIOS feature. According to manpage, one can check which of the two flash chips is currently selected like this:

**flashrom -p internal:dualbiosindex**

This error message is shown instead:

```
Unhandled programmer parameters: dualbiosindex
Aborting.
Error: Programmer initialization failed.
```

If I don't specify the **dualbiosindex** parameter this message is shown:

```
Found Programmer flash chip "Opaque flash chip"
(16384 kB, Programmer-specific) mapped at physical address
0x0000000000000000.
No operations were specified.
```

I'm also able to dump a valid copy of the firmware without problems, but I'm unable to control which of the two flash chips are being read (or written).

Superiotool from coreboot did not detect the Super I/O chip, however sensors-detect found this:

```
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8686E Super IO Sensors'                        Success!
    (address 0xa40, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```
**flashrom -V -p internal:dualbiosindex**

```
flashrom 1.2 on Linux 4.19.164_1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.7.0, GCC 9.3.0, little endian
Command line (3 args): flashrom -V -p internal:dualbiosindex
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing internal programmer
/sys/class/mtd/mtd0 does not exist
No coreboot table found.
Using Internal DMI decoder.
DMI string chassis-type: "Desktop"
DMI string system-manufacturer: "Gigabyte Technology Co., Ltd."
DMI string system-product-name: "H270N-WIFI"
DMI string system-version: "Default string"
DMI string baseboard-manufacturer: "Gigabyte Technology Co., Ltd."
DMI string baseboard-product-name: "H270N-WIFI-CF"
DMI string baseboard-version: "x.x"
Found ITE Super I/O, ID 0x8686 on port 0x2e
Found chipset "Intel H270" with PCI ID 8086:a2c4.
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... BIOS_SPI_BC = 0xa88: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
Top Swap: not enabled
SPI Read Configuration: prefetching enabled, caching enabled,
BIOS_CNTL = 0x89: BIOS Lock Enable: disabled, BIOS Write Enable: enabled
SPIBAR = 0x00007fbbfbbcb000 (phys = 0xfe010000)
0x04: 0xe004 (HSFS)
HSFS: FDONE=0, FCERR=0, AEL=1, SCIP=0, PRR34_LOCKDN=0, WRSDIS=0, FDOPSS=1, FDV=1, FLOCKDN=1
SPI Configuration is locked down.
Reading OPCODES... done
0x06: 0x3f00 (HSFC)
HSFC: FGO=0, HSFC=0, WET=0, FDBC=63, SME=0
0x0c: 0x00000000 (DLOCK)
DLOCK: BMWAG_LOCKDN=0, BMRAG_LOCKDN=0, SBMWAG_LOCKDN=0, SBMRAG_LOCKDN=0,
       PR0_LOCKDN=0, PR1_LOCKDN=0, PR2_LOCKDN=0, PR3_LOCKDN=0, PR4_LOCKDN=0,
       SSEQ_LOCKDN=0
0x50: 0x0000ffff (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0xff, BRRA 0xff
0x54: 0x00000000 FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-write.
0x58: 0x0fff0200 FREG1: BIOS region (0x00200000-0x00ffffff) is read-write.
0x5C: 0x01ff0003 FREG2: Management Engine region (0x00003000-0x001fffff) is read-write.
0x60: 0x00020001 FREG3: Gigabit Ethernet region (0x00001000-0x00002fff) is read-write.
0xa0: 0x90 (SSFS)
SSFS: SCIP=0, FDONE=0, FCERR=0, AEL=1
0xa1: 0xfe0000 (SSFC)
SSFC: SCGO=0, ACS=0, SPOP=0, COP=0, DBC=0, SME=0, SCF=6
0xa4: 0x0000     (PREOP)
0xa6: 0x0000     (OPTYPE)
0xa8: 0x00000000 (OPMENU)
0xac: 0x00000000 (OPMENU+4)
0xc4: 0xf3d82005 (LVSCC)
LVSCC: BES=0x1, WG=1, WSR=0, WEWS=0, EO=0x20, VCL=1
0xc8: 0x20452045 (UVSCC)
UVSCC: BES=0x1, WG=1, WSR=0, WEWS=0, EO=0x20
Enabling hardware sequencing because some important opcode is locked.
OK.
Unhandled programmer parameters: dualbiosindex
Aborting.
Error: Programmer initialization failed.
Restoring PCI config space for 00:1f:5 reg 0xdc
```

Any ideas why this happens?

Thanks.

vansoest, 2021-07-03:
Got the same issue and fixed it: https://review.coreboot.org/c/flashrom/+/56055
Maybe the same way will also work for you, just start `flashrom` with more verbose output `-VVV`

melvyn2, 2021-10-30:
I was also able to access both flash chips with a similar patch for the GIGABYTE Z390 AORUS MASTER:
```diff
diff --git a/it87spi.c b/it87spi.c
index 5f6fb65..bd2f7bc 100644
--- a/it87spi.c
+++ b/it87spi.c
@@ -449,6 +449,7 @@ int init_superio_ite(void)
 		case 0x8705:
 			ret |= it8705f_write_enable(superios[i].port);
 			break;
+		case 0x8688:
 		case 0x8716:
 		case 0x8718:
 		case 0x8720:
```

zailawee, 2022-05-11:
@vansoest That way also worked for me. Finally got around to push the patch: https://review.coreboot.org/c/flashrom/+/64254

closed 2022-07-31 by zailawee