flashrom/flashrom #185: [$100 Bounty] Add write-protect support (half-working patch included)

ln2max, 2021-01-13:
Some chips require certain bits (and similar features) be set before the hardware write-protect pins can be used. Flashrom doesn't currently support setting these bits, which makes it impossible to e.g hardware write-protect a BIOS chip after flashing Coreboot. I would like to see Flashrom add support for this.

Attached is a patch which was written by someone else (they can identify themselves if they like) to toggle the relevant bits for a particular flash chip, each time flashrom is run, and tell the user what's going on.

Your task, if you want to claim the bounty, is to:
1. Add some UI sugar so that the user can choose to set the appropriate bits (or not), and any other "features" needed to enable hardware write-protect, via the command line
1. Ensure the necessary bits and features can be set for a specified set of chips (see below), so Flashrom supports hardware write protect for at least all of the listed chips, and
1. Get your pull request accepted and merged (it may be wise to first check with the maintainers that they are willing to add this feature)

If $100 isn't enough to make this attractive, I am open to increasing the bounty. I am also happy to escrow it if need be. Payment will be in Bitcoin.

### Patch
[patch.zip](https://github.com/flashrom/flashrom/files/5809788/patch.zip)

### Required chips
_Not all of these may require setting bits in order to activate hardware write protection, please check the datasheets. Depending on the device, the necessary bits may be called things like Status Register Block Protect, Status Register Protect, SRWD, BP0 / BP1 / etc..._
|Vendor|Device| size|
|--|--|--|
|Eon | EN25QH32 | 4M|
|Eon| EN25QH64 | 8M|
|Macronix|MX25L3206E - MX25L3206E/MX25L3208E|4M|
|Macronix|MX25L6405,MX25L6405D, MX25L6406E/MX25L6408E, MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E/MX25L6473F, MX25L6495F|8M|
|Micron/Numonyx/ST|N25Q032..1E,  N25Q032..3E|4M|
|Micron/Numonyx/ST| N25Q064..1E,  N25Q064..3E|8M|
|Winbond | W25Q32.V, W25Q32.W | 4M|
|Winbond | W25Q64.V, W25Q32.W  | 8M|

a-m-joseph, 2021-07-05:
Here, have a thing of stuff: 
[0001-implement-wp-enable-disable-for-gigadevice-flash-chi.patch.gz](https://github.com/flashrom/flashrom/files/6762410/0001-implement-wp-enable-disable-for-gigadevice-flash-chi.patch.gz).  It's pretty lame that github rejects attachments simply because they use filenames ending in ".patch".

Minimally tested with a GD25LQ128C/GD25LQ128D in a corebooted KGPE-D16 using "-p internal":

```
# ./flashrom -p internal -w good.rom
flashrom v1.2-425-ge06f126 on Linux 5.10.10 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xb7ce0000.
Found chipset "AMD SB7x0/SB8x0/SB9x0".
Enabling flash write... OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, SPI) mapped at physical address 0x00000000ff000000.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.

# ./flashrom -p internal -c 'GD25LQ128C/GD25LQ128D' --wp-range=0,0x1000000
flashrom v1.2-425-ge06f126 on Linux 5.10.10 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xb7ce0000.
Found chipset "AMD SB7x0/SB8x0/SB9x0".
Enabling flash write... OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, SPI) mapped at physical address 0x00000000ff000000.

# ./flashrom -p internal -c 'GD25LQ128C/GD25LQ128D' --wp-enable
flashrom v1.2-425-ge06f126 on Linux 5.10.10 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xb7ce0000.
Found chipset "AMD SB7x0/SB8x0/SB9x0".
Enabling flash write... OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, SPI) mapped at physical address 0x00000000ff000000.

# ./flashrom -p internal -c 'GD25LQ128C/GD25LQ128D' --wp-status
flashrom v1.2-425-ge06f126 on Linux 5.10.10 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xb7ce0000.
Found chipset "AMD SB7x0/SB8x0/SB9x0".
Enabling flash write... OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, SPI) mapped at physical address 0x00000000ff000000.
WP: status: 0x00bc
WP: status.srp0: 1
WP: status.srp1: 0
WP: write protect is enabled.
WP: write protect range: start=0x00000000, len=0x01000000

# ./flashrom -p internal -c 'GD25LQ128C/GD25LQ128D' -E
flashrom v1.2-425-ge06f126 on Linux 5.10.10 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xb7ce0000.
Found chipset "AMD SB7x0/SB8x0/SB9x0".
Enabling flash write... OK.
Found GigaDevice flash chip "GD25LQ128C/GD25LQ128D" (16384 kB, SPI) mapped at physical address 0x00000000ff000000.
Unsetting lock bit(s) failed.
Erasing and writing flash chip... FAILED at 0x00000000! Expected=0xff, Found=0x5f, failed byte count from 0x00000000-0x00000fff: 0xdec
ERASE FAILED!
...
```

I also tested this on a GD25**B**128B -- note that this chip does not have a WP# pin at all!  There is no hardware write protect!  The goggles!!!  Only until-the-next-reboot protection.  You need a GD25**Q**128B for true hardware write protect.  It must be really handy that those two parts cannot be distinguished by software (same model number registers)... sneaky bad people can swap one for the other and not have to worry about the change being detectable electronically or causing any kind of software malfunction.

@ln2max, I think instead of "WP pin to Vcc" you meant to write "WP# pin to **GND**".  They really should have named this pin "Ignore Protection Bits" (active high).

a-m-joseph, 2021-07-05:
Also, please note that every flashrom operation, **including "verify",** will actually attempt to modify the lock bits.  In particular it will try to clear them.  So you absolutely must not use flashrom at all between setting the lock bits and shorting the WP# pin.  If you do, you'll unset the lock bits.

ln2max, 2021-08-03:
Just saw this (my email client misfiled it with the flashrom email list
traffic)... will take a look shortly!

ln2max, 2021-08-03:
Here is another flashrom patch implementing this feature which never got
merged (!):
https://mail.coreboot.org/pipermail/flashrom/2016-July/014717.html
https://mail.coreboot.org/pipermail/flashrom/2016-July/014737.html
https://mail.coreboot.org/pipermail/flashrom/2016-July/014738.html

overarching writeup here:
https://blogs.coreboot.org/blog/2016/08/26/gsoc-sr-wp-otp-wrap-up-12/

see also the author's other posts on the subject:
https://blogs.coreboot.org/blog/author/hatim/

It looks like the best way forward may just be to update and polish
Hatim's patches to work with current Flashrom, and then just keep riding
the core developers until the results get merged. 

Since it sounds like this may require a concerted campaign (emailing
people privately, poking them on social media, etc) it seems only fair
to double the bounty to $400.

Also, for general reference, the following issues are also potentially
relevant to this:
https://github.com/osresearch/heads/issues/39
https://github.com/osresearch/heads/issues/12

ln2max, 2021-08-03:
> I also tested this on a GD25**B**128B -- note that this chip does not have a WP# pin at all! There is no hardware write protect! The goggles!!! Only until-the-next-reboot protection. You need a [...]

I think if someone is in a position to physically swap out the chip, they can also just temporarily desolder the bridge to ground, write in their malicious firmware, and solder it back again. Defending against that sort of attack is why you put glitter on your laptop screws or something similar.

> @ln2max, I think instead of "WP pin to Vcc" you meant to write "WP# pin to **GND**". They really should have named this pin "Ignore Protection Bits" (active high).

I've run into chips where you actually do have to bridge the WP pin to Vcc, not to ground.

ln2max, 2021-08-03:
> Minimally tested with a GD25LQ128C/GD25LQ128D in a corebooted KGPE-D16 using "-p internal":

I don't have a Gigadevice / GD* chip to test this on, unfortunately. (And it's not in the table I gave above, in any case!)

ln2max, 2021-08-03:
> Also, please note that every flashrom operation, **including "verify",** will actually attempt to modify the lock bits. In particular it will try to clear them. So you absolutely must not use [...]

This is really annoying. Perhaps the patch should disable this behavior and stop clearing lock bits unless the user is specifically writing without write-protect.

a-m-joseph, 2021-08-03:
> > sneaky bad people can swap one for the other and not have to worry about the change being detectable electronically or causing any kind of software malfunction.

> I think if someone is in a position to physically swap out the chip, they can also just temporarily desolder the bridge to ground, write in their malicious firmware, and solder it back again. Defending against that sort of attack is why you put glitter on your laptop screws or something similar.

No no, the attack here is that the chip is swapped before the victim buys the laptop.  They buy it, wipe the drive, flash the SPI chip, and then bridge the WP# pin, incorrectly believing that this is giving them some level of protection.

This would actually be a pretty plausible attack for hardware that is no longer in production, like those Thinkpads people use coreboot on.  Buy a pile of them, install the ignores-writeprotect chip, then sell them at a small loss on ebay.  You'll be the lowest-price seller and get all the buyers.

a-m-joseph, 2021-08-03:
> Also, please note that every flashrom operation, **including "verify",** will actually attempt to modify the lock bits. In particular it will try to clear them. So you absolutely must not use [...]

> This is really annoying. Perhaps the patch should disable this behavior and stop clearing lock bits unless the user is specifically writing without write-protect.

I agree.

ln2max, 2021-08-03:
On Tue, Aug 03, 2021 at 04:33:40PM +0000, a-m-joseph wrote:
>No no, the attack here is that the chip is swapped before the victim buys the laptop.  They buy it, wipe the drive, flash the SPI chip, and then bridge the WP# pin, incorrectly believing that this is giving them some level of protection.

I think anyone who's in a position to mess with your hardware before you
get it has a lot of other options for even sneakier, nastier things to
do to you. (see, for example, the infamous Bloomberg Supermicro/"Grain
of Rice" article)

a-m-joseph, 2021-08-04:
The Bloomberg story requires custom chip manufacturing.  And quite advanced IC packaging as well.

What shocked me about the SPI flash situation is that the attack chip is already available, off the shelf, from multiple manufacturers!

ln2max, 2021-08-05:
On Wed, Aug 04, 2021 at 02:06:41PM -0700, a-m-joseph wrote:
>The Bloomberg story requires custom chip manufacturing.  And quite advanced IC packaging as well.
>
>What shocked me about the SPI flash situation is that the attack chip is already available, off the shelf, from multiple manufacturers!

Yeah, ages ago someone with much more security experience than I have
explained that it's a common error to think "one tool" or "one fix" will
keep you secure. That's almost never the case, and certainly not true
here. What this Flashrom write-protect project will (hopefully) achieve
is to block off one specific route of attack, so that the attacker has
do something more involved, something which you can (hopefully) defend
against more easily.

tlaurion, 2021-08-05:
Some input from other forks:
Adesto support: https://github.com/3mdeb/flashrom/tree/adesto_support
Adesto PoC: https://asciinema.org/a/265719

ln2max, 2021-08-05:
Nice find!

ln2max, 2021-08-11:
@pietrushnic is this something @3mdeb would be interested in?

pietrushnic, 2021-08-12:
@ln2max let us think how much effort it wold require from 3mdeb, but I assume this bount would be too small. If we will do it it will require sponsorship from 3mdeb side.

pietrushnic, 2021-08-12:
@ln2max we think this effort may contribute to KGPE-D16 revival, so maybe joining bounty to that project will increase awareness and need for such features. Feel free to watch [this issue](https://github.com/osresearch/heads/issues/719) because we will move it forward soon and planning to have a community call to coordinate the work.

ln2max, 2021-08-16:
@pietrushnic Thanks! I've left a note at that issue about joining the bounty.

ln2max, 2021-10-17:
According to @i-c-o-n on Libera/#flashrom, `nartemiev` is currently working on implementing this feature (or some variant of it).

pietrushnic, 2021-10-17:
@i-c-o-n not at this point, maybe @miczyg1 tried. We hoped that rising topic at coreboot leadership is the correct way to do. I think we have to sent some info to flashrom mailing list, but obviously we cannot track all possible channels where someone may trying to do something with flashrom that potentially overlap with mentioned effort. We would appreciate if someone can point `nartemiev` to this issue.

ln2max, 2021-10-18:
On Mon, Oct 18, 2021 at 02:29:45AM -0700, Nico Huber wrote:
>I wonder if this issue tracker could be integrated into the mailing list? From the projects perspective, GitHub is only used to mirror the repository. IMO, that creates a lot of confusion. Or maybe we should just get rid of the GitHub account.

I've sent a note to the Flashrom list notifying it that there is a lot
of discussion happening here about write-protect support.

Github issues are logically better places for discussions like this
since they bring together posts on a given issue over a long period of
time. Mailing list threads tend to get broken up as time passes.

Ideally we would have some kind of bridge between the two, so that
issue-related mailing list discussions are tracked over time. But I
think that is a separate project.

ln2max, 2021-10-19:
@pietrushnic thank you for the reference to PR #60, I've added it to the list of prior implementations.

nartemiev, 2021-11-01:
Hi @SergiiDmytruk, thanks for asking!

1. I don't have a complete list, but I plan to add support for many of chips [here](https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/5cc63e2fcb2f173accb6499404f26e14d554faf0/tests/wp_chip_info.c), though there could be some chips on that list that aren't able to be properly supported. From what I've seen, the KGPE-D16 uses 16Mb Winbond chips, do you know the specific models?

2. One goal of the rewrite was to make the writeprotect interface more programmatic, which would help with implementing the sort of functionality you describe. I hadn't considered skipping protected regions before, but it does sound useful and would probably be good a good thing to discuss with upstream flashrom maintainers as well.

ln2max, 2021-11-01:
On Sun, Oct 31, 2021 at 06:36:45PM -0700, Nikolai Artemiev wrote:
> 1. I don't have a complete list, but I plan to add support for many of chips [...]

Nikolai, in the event you overlooked it, there is an (additional) list
of chips worth supporting in the topic post for this issue.

SergiiDmytruk, 2021-11-01:
@nartemiev Thanks for the list of chips.

> From what I've seen, the KGPE-D16 uses 16Mb Winbond chips, do you know the specific models?

It seems to be common to replace them, because size is too small:

https://www.raptorengineering.com/content/kb/1.html
> the factory-installed 2MB ROM is too small. 

https://libreboot.org/docs/hardware/kgpe-d16.html#flashchips

> 2MiB flash chips are included by default, on these boards. It’s on a P-DIP 8 slot (SPI chip). The flash chip can be upgraded to higher sizes: 4MiB, 8MiB or 16MiB. With at least 8MiB, you could feasibly fit a compressed linux+initramfs image (BusyBox+Linux system) into CBFS and boot that, loading it into memory.
> 
> Libreboot has configs for 2, 4, 8 and 16 MiB flash chip sizes (default flash chip is 2MiB).

I'll ask in https://github.com/osresearch/heads/issues/719 about what people want or are using already. 

***

> I hadn't considered skipping protected regions before, but it does sound useful and would probably be good a good thing to discuss with upstream flashrom maintainers as well.

OK, below are some other things I'm considering from 3mdeb side, I wasn't trying you to make you do more than you've planned :). I think your patch looks good for our purposes and seems to not require any specific changes. However, there are several usability things from comments here and one idea taken from Hatim's patches:
* handling write-protected regions, as mentioned
* support for chips from this issue that you don't cover
* detection of hardware write-protection (when disabling software protection fails) to know when we need to work around it in code (minor thing, but currently failure to set WP will print an error on the screen)
* stop `flashrom` from removing write protection at every operation or make it restore previous state at the end, otherwise it's easy to drop protection accidentally after flashing, but before doing something physically with WP# pin
* use new WP code instead of `flashchip->unlock()` for unlocking where available

tlaurion, 2021-11-01:
> Hi @SergiiDmytruk, thanks for asking!
>
> 1. I don't have a complete list, but I plan to add support for many of chips [...]

W25Q128FVSG SOP8 chip soldered on top of a SOP8 SO8 SOIC8 TO DIP8 adapter is what I've used for initial port on Heads

SergiiDmytruk, 2021-11-03:
@nartemiev Changes listed by me above weren't deemed important enough to concentrate on. Instead, in addition to obvious testing/debugging/bugfixing for WP, 3mdeb would like to do the following:
 * implement WP/OTP in `dummy` programmer so the new code can be tested (Hatim's patches had this in feedback and I think nothing has changed since then)
 * rebase and upstream Hatim's OTP patches

Is this OK with you, so we don't end up duplicating work or otherwise conflicting? Potentially upstreaming WP can somewhat depend on tests though, but it might also be simpler to add WP and OTP to `dummy` at the same time.

nartemiev, 2021-11-03:
@SergiiDmytruk The tasks you mentioned shouldn't create conflicts with the work I have planned right now, although it may be easier to do them after the initial WP support lands.

Also, you're probably already aware, but hatim's patches were written several years ago and I'd expect rebasing them to require a lot of changes to fit in with current flashrom code.

ln2max, 2021-11-04:
On Mon, Nov 01, 2021 at 03:16:47PM -0700, SergiiDmytruk wrote:
> * detection of hardware write-protection (when disabling software protection fails) to know when we need to work around it in code (minor thing, but currently failure to set WP will print an error [...]

The process of setting hardware WP can be confusing: earlier in this
thread there was a debate about tying the WP# pin to Vcc or GND, and
many users are not experts at soldering.

It would therefore be useful to have a way to verify whether or not
hardware WP is actually enabled, without actually overwriting the chip
if the user's attempt to enable WP did not work.

blasko-coder, 2021-11-11:
I found the solution can i have 400 dollars?

Buy a Raspberry Pi 4 computer module and a io board there's a jumper that protects the cm4 from writing

SergiiDmytruk, 2021-11-12:
> It would therefore be useful to have a way to verify whether or not hardware WP is actually enabled, without actually overwriting the chip if the user's attempt to enable WP did not work.

Nikolai's patches don't do unlocking, so one will still be able to do `--wp-status`/`--wp-disable` to see if hardware protection is active. It's already existing read/verify/write operations which do unlocking unconditionally. Maybe chips with WP infrastructure support should leave it to the user instead (by setting `.unlock` to `NULL`).

blasko-coder, 2021-11-12:
sorry but I was pretty cryptic. I was writing from a smartphone...

yes take a Rapberry pi4 compute module (any RAM or wifi option) and put it on an IO board ( there a lot of makers, try the official one made by Raspberry).

There`s a jumper on the mother board that protects the compute module from writing.

I didn`t try, but you should be able to change the configuration if you do not set the write protects bits, but it cannot be flashed.

Easy and fast.

I don`t know the original intention of who started the post, but this is the solution.

If one of you could test this, would be usfull, right now I`m working on other projects.

(sometime the best solution is also the simple one)

of course I could be wrong, but for a server or a stack of servers, you`re pretty safe, the rest of the work should be done by a decent firewall

Have a good day

ln2max, 2021-11-12:
On Fri, Nov 12, 2021 at 10:37:38AM -0800, blasko-coder wrote:
> sorry but I was pretty cryptic. I was writing from a smartphone...
>
> yes take a Rapberry pi4 compute module (any RAM or wifi option) and put it on an IO board ( there a lot of makers, try the [...]

Thanks for telling us about this. It's definitely a solution if someone
is merely looking for a computing platform with very limited persistent
state.

However, adding all this functionality to Flashrom is about much more
than that. For example, the #badUSB case example: if you have a USB
device with an SPI flash chip, and you want to ensure the firmware is
not modified during operation, you can't simply tell the USB device
manufacturer to use a Raspberry Pi instead.

blasko-coder, 2021-11-13:
yes but it is not over, because as have you said before there`s a so called "high level" hacking, such as supply chain hacking, before the product is delivered to you someone could put their hands on the chip.

How to defend yourself from such thing?

I think you have to trust your security partners 100%.

you know like people bringing you gold or even more in those days

blasko-coder, 2021-11-13:
the fact is that nowadays there`s people putting viruses also in graphic cards, but in this case GPU is integrated into the SOC. I don`t if it would be possible to put some sort of virus in the firmware of the GPU, since it is loaded from the sdcard.

But I`m sure that the file is loaded from the sd card is just a backend for somewhat that is stored on the GPU itself, really I would have to check the documentation, but could be that the Foundation is hiding to us something?

ln2max, 2021-11-13:
On Sat, Nov 13, 2021 at 01:53:43AM -0800, blasko-coder wrote:
> the fact is that nowadays there`s people putting viruses also in graphic cards, but in this case GPU is integrated into the SOC. I don`t if it would be possible to put some sort of virus in the [...]

Hi,
Thanks but this particular Github issue is about adding write-protect
support to Flashrom. Your comments are interesting but they do not
belong here. Please take them somewhere more appropriate.

blasko-coder, 2021-11-13:
yes that was just a thing that come to my mind.... sorry

ln2max, 2021-12-05:
@nartemiev @3mdeb what's the current status of the patch, is it still
being worked on or has it been forgotten?

pietrushnic, 2021-12-05:
@ln2max we never forgo things we committed to. @SergiiDmytruk @macpijan please post recent status of the issue and our plans for release.

SergiiDmytruk, 2021-12-05:
@ln2max The link to Gerrit you've posted above is still valid ([this change](https://review.coreboot.org/c/flashrom/+/58474/11)) and shows stable activity. It's just that upstreaming is slow, so please be patient. Patches are being discussed, improved and are getting approved, even though none of them were merged yet.

Things I've mentioned above (tests, emulation of WP and OTP in dummyflasher and OTP support) are also part of that chain of patches, so anyone can easily check out how these new features work by pulling the top change locally.

ln2max, 2021-12-05:
On Sun, Dec 05, 2021 at 03:30:27PM -0800, SergiiDmytruk wrote:
***@***.*** The link to Gerrit you've posted above is still valid ([this change](https://review.coreboot.org/c/flashrom/+/58474/11)) and shows stable activity. It's just that upstreaming is slow, so please be patient. Patches are being discussed, improved and are getting approved, even though none of them were merged yet.
>
>Things I've mentioned above (tests, emulation of WP and OTP in dummyflasher and OTP support) are also part of that chain of patches, so anyone can easily check out how these new features work by pulling the top change locally.

Thank you for the update and the helpful instructions!

alexeyneu, 2021-12-06:
so if attacker goes first using your code all you gotta do is to resign

SergiiDmytruk, 2022-02-28:
Progress update: first few write-protect patches were merged some hours ago, many others were approved to be merged.

macpijan, 2022-05-25:
I think this patch series was merged already: https://review.coreboot.org/q/topic:writeprotect_tests

@SergiiDmytruk Can you please confirm that's the whole series which fullflls the issue desription?

SergiiDmytruk, 2022-05-25:
WP support is finally in upstream. That topic covers only my changes, the beginning of the chain is at [this commit](https://review.coreboot.org/c/flashrom/+/58474/20). OTP is [paused](https://mail.coreboot.org/hyperkitty/list/flashrom@flashrom.org/thread/BQWDBISBNQNO3EVQZGPYXVJUE3HSSODV/) at the moment.

Current master (40892b0c08fbc8029) has WP enabled for the following chips:

Atmel
* AT25SL128A

Eon
* EN25QH128

GigaDevice
* GD25LQ128C/GD25LQ128D/GD25LQ128E
* GD25LQ64(B)
* GD25Q127C/GD25Q128C
* GD25Q256D
* GD25Q32(B)
* GD25Q64(B)

Winbond
* W25Q128.V
* W25Q128.V..M
* W25Q128.W
* W25Q128.JW.DTR
* W25Q256.V
* W25Q256JV_M
* W25Q64BV/W25Q64CV/W25Q64FV
* W25Q64JV-.Q
* W25Q64.W

Unfortunately, this doesn't cover all models listed in the description, most are actually missing.

SergiiDmytruk, 2022-07-24:
Has anyone seen datasheet for MX25L6495F? Can't find this one.

SergiiDmytruk, 2022-07-27:
I've published changes that add WP support to almost all chips listed in this issue [here](https://review.coreboot.org/q/topic:more_wp). No datasheet for MX25L6495F, so can’t add it. Also didn't add WP for XT25F16, because the chip itself doesn't seem to be supported by flashrom.

If you have access to some of the other chips, please consider testing whether `--wp-*` options work fine with them and post any results here.

ln2max, 2022-08-06:
On Wed, Jul 27, 2022 at 02:17:27PM -0700, SergiiDmytruk wrote:
> I've published changes that add WP support to almost all chips listed in this issue [here](https://review.coreboot.org/q/topic:more_wp). No datasheet for MX25L6495F, so can’t add it. Also didn't add [...]

Thank you. I've added this to my to-do list and will test when I am back
in the lab.

ln2max, 2022-08-06:
SergiiDmytruk wrote:
> I've published changes that add WP support to almost all chips listed in this issue [here](https://review.coreboot.org/q/topic:more_wp). No datasheet for MX25L6495F, so can’t add it. Also didn't add [...]

That issue links to several different commits. Different commits
sometimes contain different information. Based on this, I assume the
process for testing the code is:
- download the individual commit patch files
- apply them to the source tree
- compile flashrom

This seems somewhat unwieldy and potentially error prone (e.g if the
user unintentionally skips downloading a patch set). Is there a branch
we can check out for testing purposes?

SergiiDmytruk, 2022-08-07:
That's a set of related commits stacked on top of each other. Pick the top one and hit "Download" to get instructions on pulling the latest version of all changes as a branch.

tlaurion, 2022-09-02:
Everyone here, now would be a good time to test and report here:
- Patchset for WP protection on the following chips: https://github.com/Dasharo/flashrom/blob/dasharo-develop/Documentation/chips.md
- Instructions on how to apply WP ranges (aimed at coreboot's bootblock): https://github.com/Dasharo/flashrom/pull/7
- Detection and bypass of rewriting bootblock if WP'ed: https://github.com/Dasharo/flashrom/pull/6

Please community. Participate.

ln2max, 2022-09-02:
On Fri, Sep 02, 2022 at 10:10:32AM -0700, tlaurion wrote:
> Everyone here, now would be a good time to test and report here:
> - Patchset for WP protection on the following chips: https://github.com/Dasharo/flashrom/blob/dasharo-develop/Documentation/chips.md [...]

I agree. I've got it on my to-do list to order a set of flash chips to
test this out, but work keeps getting in the way. If anyone wants to
beat me to it, you are most welcome.

macpijan, 2022-09-21:
I just tested `MX25L3206E` and `MX25L6406E` on my x230 with external programmer. The resulsts so far are there: https://github.com/Dasharo/flashrom/pull/8

ln2max, 2022-09-26:
On Wed, Sep 21, 2022 at 02:18:45AM -0700, Maciej Pijanowski wrote:
>I just tested `MX25L3206E` and `MX25L6406E` on my x230 with external programmer. The resulsts so far are there: https://github.com/Dasharo/flashrom/pull/8

Hi Maciej,
Superb work, thank you for that.

Here are the remaining chips that we may want to test, approximately in order of priority, in order to satisfy this issue:
- EN25QH32
- EN25QH64
- N25Q032..1E
- N25Q032..3E
- N25Q064..1E
- N25Q064..3E
- W25Q32.V
- W25Q32.W
- W25Q32.W
- W25Q64.V
- MX25L3208E
- MX25L6405
- MX25L6405D
- MX25L6408E
- MX25L6445E
- MX25L6465E
- MX25L6473E
- MX25L6473F

However, it looks like I'm only able to order two of the Winbond (W25..) chips. The rest seem to be unobtanium, maybe thanks to global chip shortages.

Additional chips which we want to support [are listed
here](https://github.com/Dasharo/flashrom/blob/dasharo-develop/Documentation/chips.md). I didn't add the GD chips and don't know how important it is to have (tested and working) support for them.

Therefore, since the API which @SergiiDmytruk designed checks all the boxes and everything seems to be otherwise solid, I'm not sure we need to block this until the others have been tested. (My compliments to @SergiiDmytruk on the excellent work.)

@tlaurion do you have any opinions on this?

SergiiDmytruk, 2022-09-27:
> Therefore, since the API which @SergiiDmytruk designed checks all the boxes and everything seems to be otherwise solid, I'm not sure we need to block this until the others have been tested. (My compliments to @SergiiDmytruk on the excellent work.)

Correction: I participated, but @nartemiev did the most important part.

ln2max, 2022-09-29:
>Correction: I participated, but @nartemiev did the most important part.

I stand corrected. Kudos to @nartemiev :)

tlaurion, 2022-09-29:
> @tlaurion do you have any opinions on this?

For internally upgradeable firmwares (heads on top of coreboot), where versions of coreboot changes the bootblock, my concerns were addressed under https://github.com/Dasharo/flashrom/pull/6

The general idea here is to have WP on chips. For firmwares where locked regions are not changing (say unmaintained proprietary u-boot, or neutered ME) this will work without need of external flashing.

From my understanding, specifically for coreboot maintained platforms, the bootblock will change across coreboot versions bumps, and locking bootblock will require external flashing.

For Heads, that would mean board configurations to be locked to specific versions of coreboot so that only the BIOS region can be upgraded internally. 

For all the other use cases presented in this ticket originally, where whole chips could be WP, or specific static regions untouched by software upgrades: this is not problematic.

ln2max, 2023-08-03:
Okay, based on @tlaurion's comments, we will say this is pretty much
done.

Have all changes and patches been merged?

SergiiDmytruk, 2023-08-03:
> Have all changes and patches been merged?

I'm not aware of any pending reviews related to this issue.

tlaurion, 2023-08-26:
> Okay, based on @tlaurion's comments, we will say this is pretty much
> done.
> 
> Have all changes and patches been merged?
> 

Has someone reproduced the results from https://github.com/Dasharo/flashrom/pull/8 ?

This pr on documentation should be merged and bounty here given so this is closed.

[28 more replies]