flashrom/flashrom #190: "W25Q80EW" ERASE FAILED!

Zerotul2015, 2021-02-12:
flashrom v1.2 on Linux 5.8.0-43-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q80EW" (1024 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... FAILED at 0x00000000! Expected=0xff, Found=0x80, failed byte count from 0x00000000-0x00000fff: 0xff6
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000000! Expected=0xff, Found=0x80, failed byte count from 0x00000000-0x00007fff: 0x7faf
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000000! Expected=0xff, Found=0x80, failed byte count from 0x00000000-0x0000ffff: 0xff61
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000000! Expected=0xff, Found=0x80, failed byte count from 0x00000000-0x000fffff: 0xff3be
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000000! Expected=0xff, Found=0x80, failed byte count from 0x00000000-0x000fffff: 0xff3b9
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Apparently at least some data has changed.
Your flash chip is in an unknown state.
Please report this on IRC at chat.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org, thanks!

agitimtim, 2021-02-19:
When I get this error, removing the chip from the board and then flashing seems to work.

Zerotul2015, 2021-03-21:
> When I get this error, removing the chip from the board and then flashing seems to work.

Thanks, the advice helped.

closed 2021-03-21 by Zerotul2015