flashrom/flashrom #195: Implement support for 1a86:5523 QinHeng Electronics CH341

Kreyren, 2021-03-05:
```
Bus 003 Device 010: ID 1a86:5523 QinHeng Electronics CH341 in serial mode, usb to serial port converter
```

Kreyren, 2021-03-05:
Invalid: Requires the pin 1 + 2 connected together to show 5512

closed 2021-03-05 by Kreyren

firefoxlover, 2023-09-25:
I dont understand this progress. Is this programmer not supported at all?

Kreyren, 2023-10-03:
> I dont understand this progress. Is this programmer not supported at all? -- @firefoxlover (https://github.com/flashrom/flashrom/issues/195#issuecomment-1734520037)

It seems to be supported and works just for the board i was using it needed the pins 1 and 2 bridged together for the programmer to show up and work.

So I though that it meant that the programmer is not supported, but it was just an issue of lack of documentation and mis-configuration as a result of it.