flashrom/flashrom #196: Debugger outputting 512kB instead of 256kB?

Kreyren, 2021-03-05:
Referencing: https://github.com/kreyware/repairs/issues/1

I have CH341A trying to flash GV-RXVEGA56GAMING with VBIOS 016.001.001.000.033736 which is 256kB, but flashrom  [flashrom v1.2-204-gb05c9b1-dirty on Linux 5.10.0-3-amd64 (x86_64)] reads it as 512kB which would make it one of a kind Vega 56 and refuses to write.

Can you provide more info?

LOG: https://gist.github.com/Kreyren/3c2582bb6f01aa6cf5da8be68e3cbb41

Th3Fanbus, 2021-03-05:
Hi Kreyren,

The log says that flashrom was unable to identify a known flash chip:

> Found Generic flash chip "unknown SPI chip (REMS)" (0 kB, SPI) on ch341a_spi.

Looks like the RDID (Read Identification) command returned all ones. Since flashrom identifies nearly all SPI flash chips using the RDID values, it's not a surprise that it was unable to detect your chip.

This can happen when the connection is unreliable. To do in-circuit flashing (e.g. using a clip to flash a chip without desoldering it from the board), the board needs to have a diode (one-way door for electricity) so that powering the flash chip through the programmer does not power other components on the board. However, not all boards have this diode. Without it, in-circuit flashing may not always work reliably. If flashrom cannot reliably identify the flash chip, do not attempt to flash.

Kreyren, 2021-03-05:
> This can happen when the connection is unreliable. To do in-circuit flashing (e.g. using a clip to flash a chip without desoldering it from the board), the board needs to have a diode (one-way door for electricity) so that powering the flash chip through the programmer does not power other components on the board. However, not all boards have this diode. Without it, in-circuit flashing may not always work reliably. If flashrom cannot reliably identify the flash chip, do not attempt to flash. @Th3Fanbus 

Agree performing the flash on board made it to return random values and unreliable connection for the reading so i desoldered it and put it in a debugger with the same issue.

Setup(chip installed on green arrow, PCB provided with the debugger): 
[image]

---

Any idea what might make it to not get the RDID here? FWIW it seems that the VBIOS chip is damaged.

Kreyren, 2021-03-05:
FWIW the connection in this setup seems stable as reading the data gets me the the same file verified through `sha256sum`.

Kreyren, 2021-03-05:
I checked the connection and now i am getting: https://gist.githubusercontent.com/Kreyren/7e35e7c285bf21ca18c8af939e51f3c6/raw/81a69e2d002e95296267ff0e92ffbaf737847435/gistfile1.txt

> Found GigaDevice flash chip "GD25Q40(B)" (512 kB, SPI) on ch341a_spi.
> Found GigaDevice flash chip "GD25Q40(B)" (512 kB, SPI).

Kreyren, 2021-03-05:
Reading hexdump:

> 00040000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
> *
> 00080000

Kreyren, 2021-03-06:
```console
kreyren@leonid:~/Downloads$ dd if=/dev/zero count=512 >> Gigabyte.RXVega56.8192.190723.rom
512+0 records in
512+0 records out
262144 bytes (262 kB, 256 KiB) copied, 0.00171236 s, 153 MB/s
kreyren@leonid:~/Downloads$ ls -s Gigabyte.RXVega56.8192.190723.rom
512 Gigabyte.RXVega56.8192.190723.rom
kreyren@leonid:~/Downloads$ sudo flashrom --verbose --programmer ch341a_spi -w Gigabyte.RXVega56.8192.190723.rom
flashrom v1.2-204-gb05c9b1-dirty on Linux 5.10.0-3-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.7.0, GCC 10.2.1 20210110, little endian
Command line (5 args): flashrom --verbose --programmer ch341a_spi -w Gigabyte.RXVega56.8192.190723.rom
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing ch341a_spi programmer
Device revision is 3.0.4
...
Probing for GigaDevice GD25Q40(B), 512 kB: compare_id: id1 0xc8, id2 0x4013
Found GigaDevice flash chip "GD25Q40(B)" (512 kB, SPI) on ch341a_spi.
Chip status register is 0x00.
Chip status register: Status Register Write Disable (SRWD, SRP, ...) is not set
Chip status register: Block Protect 4 (BP4) is not set
Chip status register: Block Protect 3 (BP3) is not set
Chip status register: Block Protect 2 (BP2) is not set
Chip status register: Block Protect 1 (BP1) is not set
Chip status register: Block Protect 0 (BP0) is not set
Chip status register: Write Enable Latch (WEL) is not set
Chip status register: Write In Progress (WIP/BUSY) is not set
...
Found GigaDevice flash chip "GD25Q40(B)" (512 kB, SPI).
===
This flash part has status UNTESTED for operations: PROBE READ ERASE WRITE
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading old flash chip contents... done.
Erasing and writing flash chip... Trying erase function 0... 0x000000-0x000fff:EFAILED at 0x00000021! Expected=0xff, Found=0xfe, failed byte count from 0x00000000-0x00000fff: 0xa3e
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Trying erase function 1... 0x000000-0x007fff:EW, 0x008000-0x00ffff:EW, 0x010000-0x017fff:EW, 0x018000-0x01ffff:EW, 0x020000-0x027fff:EW, 0x028000-0x02ffff:EW, 0x030000-0x037fff:EW, 0x038000-0x03ffff:EW, 0x040000-0x047fff:W, 0x048000-0x04ffff:W, 0x050000-0x057fff:W, 0x058000-0x05ffff:W, 0x060000-0x067fff:W, 0x068000-0x06ffff:W, 0x070000-0x077fff:W, 0x078000-0x07ffff:W
Erase/write done.
Verifying flash... FAILED at 0x00017c00! Expected=0x01, Found=0xff, failed byte count from 0x00000000-0x0007ffff: 0xcf1
Your flash chip is in an unknown state.
Please report this on IRC at chat.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org, thanks!
```

As suggested in i tried to append 256 FF bytes to the rom file, but the flash still fails.

Kreyren, 2021-03-06:
Seems to erase cleanly outside of the flashing though?

```console
kreyren@leonid:~/Downloads$ sudo flashrom --programmer ch341a_spi --erase
flashrom v1.2-204-gb05c9b1-dirty on Linux 5.10.0-3-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found GigaDevice flash chip "GD25Q40(B)" (512 kB, SPI) on ch341a_spi.
===
This flash part has status UNTESTED for operations: PROBE READ ERASE WRITE
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Erasing and writing flash chip... Erase/write done.
```

Kreyren, 2021-03-06:
Now i was able to perform the flash assuming resolved.

closed 2021-03-06 by Kreyren