flashrom/flashrom #210: "BY25Q128AS"  invalid response 0x00 (to command 0x13)

eliasmagn, 2021-05-23:
@cecada @dhendrix
with compiled from master  flashrom v1.2-318-gc49aaee
`Found Boya/BoHong Microelectronics flash chip "B.25Q128AS" (16384 kB, SPI) on serprog.`

and with @cecada 's fork master = flashrom v1.2-91-gefc4e7d
`Found Boya flash chip "BY25Q128AS" (16384 kB, SPI) on serprog.`

Added support for BY25Q128AS - BoyaMicro Electronics #161

i get 
` Error: invalid response 0x00 from device (to command 0x13)`

flash is working, system is still booting, its a smal router i try to port to openwrt
same fault in or out of cirquit

tested with arduino pro micro on 3.3v
any suggestions?

eliasmagn, 2021-05-23:
i get same fault on a known working chip i have to investigate further. i will write new comment or close issue directly if i find or wont find solution . 
regards

cecada, 2021-05-24:
I added the Boya commit, but not the other changes by @dhendrix. Are using a Boya chip? What router are you using? Can you post a picture of the chip with it's markings?

closed 2025-10-22 by eliasmagn