flashrom/flashrom #220: Please add support for W25Q128JV

Baa14453, 2021-08-20:
Data sheet here: https://datasheet.lcsc.com/szlcsc/Winbond-Elec-W25Q128JVFIQ_C111478.pdf

Upload of it here: [Winbond-Elec-W25Q128JVFIQ_C111478.pdf](https://github.com/flashrom/flashrom/files/7022605/Winbond-Elec-W25Q128JVFIQ_C111478.pdf)

If I manually specify the chip `W25Q128.V` then it usually detects it (about 1/3 tries) and tries to read, but the output file is always all 0s

Baa14453, 2021-08-20:
Huh, I just tried to read it again and it now says that `Block protection is disabled.` I don't know what changed, maybe increasing the voltage helped

The firmware is still full of 0s though.
[log2.txt](https://github.com/flashrom/flashrom/files/7023166/log2.txt)

dhendrix, 2021-08-21:
In addition to ensuring voltage doesn't drop too low (the chip's datasheet gives a range of 2.7V-3.6V), try running at a reduced SPI frequency (issue #29). Both are especially impacted if you are using wires longer than about ~15cm.

Baa14453, 2021-08-21:
> In addition to ensuring voltage doesn't drop too low (the chip's datasheet gives a range of 2.7V-3.6V), try running at a reduced SPI frequency (issue #29). Both are especially impacted if you are using wires longer than about ~15cm.

Thanks for your suggestion, at 512 speed, the first time it detected it as `"unknown SPI chip (REMS)" (0 kB, SPI)`. Second time it found nothing, on the third try it found `"W25Q128.V" (16384 kB, SPI)` and attempted to read the firmware but the file is all 0s still.

At 30,000 speed, it detects `"unknown Winbond (ex Nexcom) SPI chip" (0 kB, SPI)` about every other time, and nothing otherwise. :thinking: 
The chip supports standard, Dual and Quad SPI speed but I've only hooked into the first DI and DO ports, I'm not sure if it might be trying to use dual or quad and causing problems?

My new capacitors arrive tomorrow so I'll try those, just testing different speeds now

dhendrix, 2021-08-22:
> The chip supports standard, Dual and Quad SPI speed but I've only hooked into the first DI and DO ports, I'm not sure if it might be trying to use dual or quad and causing problems?

It's shouldn't be trying to use dual or quad mode unless you made the RPi's SPI controller work in one of those modes (it probably doesn't support quad mode, not sure about dual).

Did you connect the unused pins (/HOLD and /WP) on the flash chip to Vcc?

Baa14453, 2021-08-22:
> It's shouldn't be trying to use dual or quad mode unless you made the RPi's SPI controller work in one of those modes (it probably doesn't support quad mode, not sure about dual).
>
> Did you connect [...]

no, I'll try that thanks.
Tried with the capacitors but no difference so far

Baa14453, 2021-08-22:
> no, I'll try that thanks.
> Tried with the capacitors but no difference so far

Tried it all but each time the dump is almost all 0s :(
Maybe the `W25Q128.V` and `W25Q128.JV` are different in some way?
[dvr2.zip](https://github.com/flashrom/flashrom/files/7028319/dvr2.zip)

LmarLoe, 2021-08-23:
Recently I successfully flashed W25Q128.V. I tried it on different boards. It works, but not always. Takes several attempts to make a dump. Sometimes it helps to reconnect the chip. Seems even orientation of the wires is matter. Maybe try to isolate from EMR or replace wires.

Baa14453, 2021-08-23:
> Recently I successfully flashed W25Q128.V. I tried it on different boards. It works, but not always. Takes several attempts to make a dump. Sometimes it helps to reconnect the chip. Seems even orientation of the wires is matter. Maybe try to isolate from EMR or replace wires. Chip marking: 25Q128JVSQ

ok I will keep trying, my chip marking is 25Q128JVFQ

Baa14453, 2021-08-23:
So I've hooked it back into my RaspberryPi's 3.3v pin instead of my variable PSU
It's only registering 2.2v on the voltmeter and now produces a firmware consisting of all FFs

I consider this a welcome improvement to all 00s, I will keep testing things.
[image]

dazhbog, 2023-06-24:
Sorry to bring this topic up, I am playing with this very chip and I am getting this "UNTESTED" message:

```
./flashrom.exe -V -c W25Q128.V -p serprog:dev=COM3:230400 -x
flashrom v1.3.0 on Windows 10.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 100ns).
flashrom was built with GCC 12.2.0, little endian
Command line (6 args): flashrom.exe -V -c W25Q128.V -p serprog:dev=COM3:230400 -x
Initializing serprog programmer
Baud rate is 230400.
serprog: connected - attempting to synchronize
.
serprog: Synchronized
serprog: Interface version ok.
serprog: Bus support: parallel=off, LPC=off, FWH=off, SPI=on
Warning: Automatic command availability check failed for cmd 0x08 - won't execute cmd
Warning: Automatic command availability check failed for cmd 0x11 - won't execute cmd
serprog: Programmer name is "Pulse Mem Sys"
serprog: Serial buffer size is 65535
serprog: Output drivers enabled
The following protocols are supported: SPI.
Probing for Winbond W25Q128.V, 16384 kB: compare_id: id1 0xef, id2 0x4018
Added layout entry 00000000 - 00ffffff named complete flash
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on serprog.
Chip status register is 0x00.
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading flash... Read operation failed!
FAILED.
serprog: Output drivers disabled
```

Not sure if I am missing something, I just implemented the serprog protocol on my production hardware, so not sure if I need to do something more to get it working. My chip is W25Q128FV or W25Q128JV. Happy to provide logs and stuff but not sure whats important to share

I also have the raw serial logs

```
>>> 00
<<< 06
>>> 10
<<< 15 06
>>> 10
<<< 15 06
>>> 01
<<< 06 01 00
>>> 02
<<< 06 3F 00 3D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>>> 05
<<< 06 08
>>> 12 08
<<< 06
>>> 12 08
<<< 06
>>> 03
<<< 06 Pulse Mem Sys00 00 00
>>> 04
<<< 06 FF FF
>>> 15 01
<<< 06
>>> 13 01 00 00 03 00 00 9F
<<< 06 EF 40 18
>>> 13 01 00 00 02 00 00 05
<<< 06 00 00
>>> 13 01 00 00 02 00 00 05
<<< 06 00 00
>>> 13 04 00 00 00 00 01 15 
>>> 03
<<< 06 Pulse Mem Sys00 00 00
>>> 00
<<< 06
>>> 00
<<< 06
>>> 00
<<< 06
>>> 15 00
<<< 06
```

dazhbog, 2023-06-25:
Ok, got it to play now, the 

```
>>> 13 04 00 00 00 00 01 15 
```
command wasn't actioned properly by the reader. Now its reading fine.

Kreyren, 2024-04-08:
I have W25Q128JVSIQ that i am trying to flash via ch341a_spi (black):

```console
├─ ~ via ❄️  impure (shell)
╰───❯ sudo flashrom -h -V -p ch341a_spi -w ~/Downloads/acer_nitro/BIOS_Acer_1.11_A_A/abobios.bin_extracted/BIOS-UEFI\ \[0x0012F150-0x0112F150].bin
flashrom v1.3.0 on Linux 6.7.10-xanmod1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Usage: flashrom [-h|-R|-L|
	-p <programmername>[:<parameters>] [-c <chipname>]
		(--flash-name|--flash-size|
		 [-E|-x|(-r|-w|-v) <file>]
		 [(-l <layoutfile>|--ifd| --fmap|--fmap-file <file>) [-i <region>[:<file>]]...]
		 [-n] [-N] [-f])]
	[-V[V[V]]] [-o <logfile>]

 -h | --help                        print this help text
 -R | --version                     print version (release)
 -r | --read <file>                 read flash and save to <file>
 -w | --write (<file>|-)            write <file> or the content provided
                                    on the standard input to flash
 -v | --verify (<file>|-)           verify flash against <file>
                                    or the content provided on the standard input
 -E | --erase                       erase flash memory
 -V | --verbose                     more verbose output
 -c | --chip <chipname>             probe only for specified flash chip
 -f | --force                       force specific operations (see man page)
 -n | --noverify                    don't auto-verify
 -N | --noverify-all                verify included regions only (cf. -i)
 -x | --extract                     extract regions to files
 -l | --layout <layoutfile>         read ROM layout from <layoutfile>
      --wp-disable                  disable write protection
      --wp-enable                   enable write protection
      --wp-list                     list supported write protection ranges
      --wp-status                   show write protection status
      --wp-range=<start>,<len>      set write protection range (use --wp-range=0,0
                                    to unprotect the entire flash)
      --wp-region <region>          set write protection region
      --flash-name                  read out the detected flash name
      --flash-size                  read out the detected flash size
      --fmap                        read ROM layout from fmap embedded in ROM
      --fmap-file <fmapfile>        read ROM layout from fmap in <fmapfile>
      --ifd                         read layout from an Intel Firmware Descriptor
 -i | --include <region>[:<file>]   only read/write image <region> from layout
                                    (optionally with data from <file>)
      --image <region>[:<file>]     deprecated, please use --include
 -o | --output <logfile>            log output to <logfile>
      --flash-contents <ref-file>   assume flash contents to be <ref-file>
 -L | --list-supported              print supported devices
 -p | --programmer <name>[:<param>] specify the programmer device. One of
    internal, dummy, nic3com, nicrealtek, gfxnvidia, raiden_debug_spi, drkaiser,
    satasii, atavia, it8212, ft2232_spi, serprog, buspirate_spi, dediprog,
    developerbox, rayer_spi, pony_spi, nicintel, nicintel_spi, nicintel_eeprom,
    ogp_spi, satamv, linux_mtd, linux_spi, usbblaster_spi, pickit2_spi,
    ch341a_spi, digilent_spi, stlinkv3_spi.

You can specify one of -h, -R, -L, -E, -r, -w, -v or no operation.
If no operation is specified, flashrom will only probe for flash chips.

❄️ kreyren@pelagus  21:17:12 08.04.2024 +02:00 
├─ ~ via ❄️  impure (shell)
╰───❯ sudo flashrom -V -p ch341a_spi -w ~/Downloads/acer_nitro/BIOS_Acer_1.11_A_A/abobios.bin_extracted/BIOS-UEFI\ \[0x0012F150-0x0112F150].bin
flashrom v1.3.0 on Linux 6.7.10-xanmod1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
flashrom was built with GCC 12.2.0, little endian
Command line (5 args): flashrom -V -p ch341a_spi -w /home/kreyren/Downloads/acer_nitro/BIOS_Acer_1.11_A_A/abobios.bin_extracted/BIOS-UEFI [0x0012F150-0x0112F150].bin
Initializing ch341a_spi programmer
[... 398 lines cut ...]
Probing for ST unknown ST SPI chip, 0 kB: compare_id: id1 0xef, id2 0x4018
Probing for Sanyo unknown Sanyo SPI chip, 0 kB: compare_id: id1 0xef, id2 0x4018
Probing for Winbond unknown Winbond (ex Nexcom) SPI chip, 0 kB: compare_id: id1 0xef, id2 0x4018
Probing for Generic unknown SPI chip (RDID), 0 kB: compare_id: id1 0xef, id2 0x4018
Probing for Generic unknown SPI chip (REMS), 0 kB: compare_id: id1 0xef, id2 0x17
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI).
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading old flash chip contents... done.
[... cut ...] S, 0xff1000-0xff1fff:S, 0xff2000-0xff2fff:S, 0xff3000-0xff3fff:S, 0xff4000-0xff4fff:S, 0xff5000-0xff5fff:S, 0xff6000-0xff6fff:S, 0xff7000-0xff7fff:S, 0xff8000-0xff8fff:S, 0xff9000-0xff9fff:S, 0xffa000-0xffafff:S, 0xffb000-0xffbfff:S, 0xffc000-0xffcfff:S, 0xffd000-0xffdfff:S, 0xffe000-0xffefff:S, 0xfff000-0xffffff:E
Erase/write done.
Verifying flash... VERIFIED.
```

It says that it wrote and verified the flash, but the laptop that i am trying to fix (Acer Nitro 5) is still not booting :thinking:

carlosgonz0, 2025-09-22:
Hi

I have HP Z220 bricked, because flashrom do not support  25Q128BVFG .