flashrom/flashrom #226: Could flashrom read/write a 1-Wire interface? The chip specifically is DS2434.

TheMrRandomDude, 2021-10-10:
Hello.

Chip's documentation can be read here: https://docs.rs-online.com/e82f/0900766b800accfc.pdf

The hardware connection itself should be straight forward, Vcc to Vcc, Gnd to Gnd, and data pin to mosi. [Someone already did it on this forum with a different flasher](https://forum.reveltronics.com/viewtopic.php?f=32&t=897).

The only part on which I'm stuck is software now. Not sure if flashrom can natively read the chip, as I haven't recieved it yet. Figured I'd ask. Perhaps bitbanging is the answer, just not 100% sure how to implement it.

Thanks :)

jesultra, 2023-04-23:
The DS2434 chip does not use SPI , and the bytes that are transmitted via its 1-wire bus do not in any way resemble the SPI flash protocols supported by flashrom, so while anything is possible, I do not think that it would make sense to add  support for DS2434 to flashrom.

TheMrRandomDude, 2023-04-23:
Oh boy, I posted this a long time ago. Yeah I figured that out so far, thanks for the reply. :)

For any person looking into this, the answer is with flashrom no, but the proper tool for OneWire is:

1. Raspberry Pi OneWire library
2. Digital logic analyzer to see raw bits and then use another library
3. OneWireHub on github

Closing this, as it's solved. Thanks.

closed 2023-04-23 by TheMrRandomDude