flashrom/flashrom #229: Progress bars

busterbeam, 2021-10-20:
I have quite the thing for progress bars but my `C` isn't great. put simply I tried this
```c
unsigned long numbytes = fwrite(buf, 1, size, image);
```
as this 
```c
unsigned long numbytes = 0;
	
for (unsigned long i = 0; i < size; i += 1024) {
	buf += 1024;
	numbytes += fwrite(buf, 1, 1024, image);
	printf("\x1B[2K %d %% Complete \r", ((float)i / (float)size) * 100.0f);
	fflush(stdout);
}
```

I've also added more `msg_cinfo` functions prior to it to see control flow but after `make clean` and `make`  It doesn't make a difference!!  is there some buffered output issue inside your flashrom binary??  I would appreciate someone breaking down why it just holds on `"Reading flash... "` and doesn't provide anymore info after that??

busterbeam, 2021-10-20:
I can imagine you could add a cmd argument whereby if they want read/write to have a progress bar by doing `status=progress` and `bs={block_size}` like `dd`?

vinibali, 2022-11-01:
(https://ticket.coreboot.org/issues/437)[https://ticket.coreboot.org/issues/437]

busterbeam, 2022-11-01:
cool @vinibali thanks.  I think I did make progress on this a while ago.  I do plan to close this issue myself

vinibali, 2022-11-03:
I'm really looking forward to see your solution!