flashrom/flashrom #230: flashrom not work hp 8200 sff

kocoman1, 2021-10-30:
<span style="font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.301961); -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(255, 255, 255); color: rgb(255, 255, 255); font-family: UICTFontTextStyleBody; font-size: 17px; background-color: rgb(28, 28, 30); float: none; display: inline !important;">Hi i have a hp 8200 sff with hp bios. i am trying to flash coreboot on it but it fails with this log.  any ideas? thanks</span><div dir="ltr" style="font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.301961); -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(255, 255, 255); color: rgb(255, 255, 255); font-family: UICTFontTextStyleBody; font-size: 17px;"></div><div style="font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.301961); -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(255, 255, 255); color: rgb(255, 255, 255); font-family: UICTFontTextStyleBody; font-size: 17px;"><br></div><div style="font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.301961); -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(255, 255, 255); color: rgb(255, 255, 255); font-family: UICTFontTextStyleBody; font-size: 17px;"><google-sheets-html-origin style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-family: -webkit-standard; -webkit-text-size-adjust: auto;">

flashrom -p internal -c "MX25L6406E/MX25L6408E" --ifd -i bios -w build/coreboot.rom
--
flashrom v1.2 on Linux 5.13.0-19-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org
 
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Q67".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... Warning: Setting Bios Control at 0xdc from 0x02 to 0x01 failed.
New value is 0x02.
SPI Configuration is locked down.
The Flash Descriptor Override Strap-Pin is set. Restrictions implied by
the Master Section of the flash descriptor are NOT in effect. Please note
that Protected Range (PR) restrictions still apply.
PR0: Warning: 0x007f0000-0x007fffff is read-only.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
OK.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
Reading ich descriptor... done.
Peculiar firmware descriptor, assuming Ibex Peak compatibility.
Using region: "bios".
Reading old flash chip contents... done.
Erasing and writing flash chip... Transaction error!
spi_write_cmd failed during command execution at address 0x510000
Reading current flash chip contents... done. Looking for another erase function.
spi_write_cmd failed during command execution at address 0x510000
Reading current flash chip contents... done. Looking for another erase function.
spi_write_cmd failed during command execution at address 0x510000
Reading current flash chip contents... done. Looking for another erase function.
spi_simple_write_cmd failed during command execution
Reading current flash chip contents... done. Looking for another erase function.
spi_simple_write_cmd failed during command execution
Reading current flash chip contents... done. Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Good, writing to the flash chip apparently didn't do anything.
This means we have to add special support for your board, programmer or flash
chip. Please report this on IRC at chat.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org, thanks!
-------------------------------------------------------------------------------
You may now reboot or simply leave the machine running.

</google-sheets-html-origin></div>

tizbac, 2023-04-23:
Same problem me too

kocoman1, 2023-04-23:
there was a option no-verify-all or something.. but its the hp bios having hidden write protection

tizbac, 2023-04-24:
Yeah, ended up flashing with Ch341a everything worked

kerta1n, 2023-12-06:
@tizbac how did you get the ch341a to work? I'm trying without the 3.3>1.8v stepdown, but nothing seems to work. I get, 
`dietpi@lepotato:/tmp$ sudo flashrom --programmer ch341a_spi -w seabios_withgrub_hp8200sff_8mb_libgfxinit_txtmode_usqwerty_THEONEYOUNEED.rom
flashrom v1.2 on Linux 6.1.11-meson64 (aarch64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Couldn't open device 1a86:5512.
Error: Programmer initialization failed.`