flashrom/flashrom #231: flashrom hangs with latest buspirate firmware

nhorman, 2021-12-01:
testing with buspirate firmware at commit 4bd112bfefbc6ec07e56b944bd2753801ffb558e, trying to program a winbond 32Mb flash chip.

The flashrom utility hangs when sending this command sequence to the buspirate:

0x4 0x0 0x1 0x0 0x0 0x6
 This is a buspirate read/write command writing a single byte to the chip over the spi bus, specifically the write enable command (0x6).  buspirate_sendrecv indicates we are writing 6 bytes and receiving 1 (which is more than is the command string due to the +1 in the call made from buspirate_spi_send_command_v2 in flashrom

I _think_ I know whats going on, but was hoping someone else could take a closer look.

Flashrom sends the above command string to indicate that the spi bus should send back 0 bytes to the buspirate, but it adds one to the receive count because it expects the bus pirate to send back a success/failure code (0 or 1), in flashrom, this addition of a single byte to read outside the command string, causes it to sit on a read call for the serial port

meanwhile in the latest buspirate firmware, in spi_read_write_io, we fetch the bytes_to_read from the command string (which is 0).  As such, at the bottom of this function, we have this code:
if (bytes_to_read > 0) {
    spi_read_to_uart(bytes_to_read);
  }

so spi_read_write_io is expecting the status byte to be included in the read count, whereas buspirate_spi_send_command_v2  is not.

I  _think_ the right solution is to modify buspirate_spi_send_command_v2 so that readcnt is incremented in the command strings to include the status byte, which triggers the latest buspirate firmware to send said status byte, but I wanted to check here first, as I wasn't sure if there were any other repercussions to making that change.

BlazejCwik, 2024-03-24:
Hi, any progress with that?

redawl, 2024-04-22:
Running into this as well, I will take a look, but I am not familiar with flashrom so it could take a while

nhorman, 2024-04-22:
I've moved away from this, but FWIW, modifying buspirate_spi_send_command_v2 as I described above got me working again.  Just wasn't sure if it was the right fix or not

redawl, 2024-04-22:
For others who are running into this issue, see this issue on the the Bus Pirate repo:
https://github.com/BusPirate/Bus_Pirate/issues/131

The firmware version they mention worked for me. Seems to be a problem on the Bus_Pirate firmware end.

jacobcayetano1, 2024-10-31:
@nhorman Would you mind describing the change in further detail or posting your modification? It is not immediately clear to me where you need to include the status byte.